firesim/firesim #69: Non-specific debug output

nullromo, 2018-08-15:
Output from running `infrasetup` when there is a chisel compile-time error does not display any information about the error itself.

davidbiancolin, 2018-09-05:
What was in the log file?

nullromo, 2018-09-05:
The logfile contained the same information that the stdout messages said.
As an example, I added the line

`var`

to the middle of `/sim/src/main/scala/endpoints/BlockDevWidget.scala` and then ran `firesim infrasetup`.

Output:
```
FireSim Manager. Docs: http://docs.fires.im
Running: infrasetup

Building FPGA software driver for FireSimNoNIC-FireSimRocketChipQuadCoreConfig-FireSimDDR3FRFCFSLLC4MBConfig
Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 300, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 249, in main
    globals()[args.task](simconf)
  File "/home/centos/firesim/deploy/firesim", line 66, in infrasetup
    runtime_conf.infrasetup()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 240, in infrasetup
    self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 293, in infrasetup_passes
    self.pass_build_required_drivers()
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 276, in pass_build_required_drivers
    server.get_server_hardware_config().build_fpga_driver()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 110, in build_fpga_driver
    localcap = local("""make DESIGN={} TARGET_CONFIG={} PLATFORM_CONFIG={} f1""".format(design, target_config, platform_config), capture=True)
  File "/usr/lib64/python2.7/site-packages/fabric/operations.py", line 1239, in local
    error(message=msg, stdout=out, stderr=err)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 358, in error
    return func(message)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 62, in abort
    raise e
SystemExit: 1
The full log of this run is:
/home/centos/firesim/deploy/logs/2018-09-05--21-03-19-infrasetup-GD5CNBBYVG0WGN1O.log
```

And `cat /home/centos/firesim/deploy/logs/2018-09-05--21-03-19-infrasetup-GD5CNBBYVG0WGN1O.log` shows:
```
2018-09-05 21:03:19,468 [main        ] [INFO ]  FireSim Manager. Docs: http://docs.fires.im
Running: infrasetup

2018-09-05 21:03:19,468 [__init__    ] [DEBUG]  {'hwconf_dict': {'fireboom-singlecore-no-nic-ddr3-llc4mb': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1ca28>,
                 'fireboom-singlecore-no-nic-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1cb00>,
                 'firesim-quadcore-nic-ddr3-llc4mb': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1cbd8>,
                 'firesim-quadcore-nic-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1cb90>,
                 'firesim-quadcore-no-nic-ddr3-llc4mb': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1cab8>,
                 'firesim-quadcore-no-nic-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1ca70>,
                 'firesim-singlecore-nic-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1cc20>,
                 'firesim-singlecore-no-nic-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7f2531a1cb48>}}
2018-09-05 21:03:19,470 [__init__    ] [DEBUG]  {'defaulthwconfig': 'firesim-quadcore-no-nic-ddr3-llc4mb',
 'f1_16xlarges_requested': 0,
 'f1_2xlarges_requested': 1,
 'linklatency': 6405,
 'm4_16xlarges_requested': 0,
 'netbandwidth': 200,
 'no_net_num_nodes': 1,
 'run_instance_market': 'ondemand',
 'runfarmtag': 'mainrunfarm',
 'spot_interruption_behavior': 'terminate',
 'spot_max_price': 'ondemand',
 'switchinglatency': 10,
 'terminateoncompletion': False,
 'topology': 'no_net_config',
 'workload_name': 'linux-uniform.json'}
2018-09-05 21:03:19,470 [get_deploytr] [DEBUG]  Setting deploytriplet by querying the AGFI's description.
2018-09-05 21:03:19,470 [get_afi_for_] [DEBUG]  agfi-00f8511c144ca7774
[... 9 lines cut ...]
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 249, in main
    globals()[args.task](simconf)
  File "/home/centos/firesim/deploy/firesim", line 66, in infrasetup
    runtime_conf.infrasetup()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 240, in infrasetup
    self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 293, in infrasetup_passes
    self.pass_build_required_drivers()
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 276, in pass_build_required_drivers
    server.get_server_hardware_config().build_fpga_driver()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 110, in build_fpga_driver
    localcap = local("""make DESIGN={} TARGET_CONFIG={} PLATFORM_CONFIG={} f1""".format(design, target_config, platform_config), capture=True)
  File "/usr/lib64/python2.7/site-packages/fabric/operations.py", line 1239, in local
    error(message=msg, stdout=out, stderr=err)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 358, in error
    return func(message)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 62, in abort
    raise e
SystemExit: 1
2018-09-05 21:03:46,722 [<module>    ] [INFO ]  The full log of this run is:
/home/centos/firesim/deploy/logs/2018-09-05--21-03-19-infrasetup-GD5CNBBYVG0WGN1O.log
```

So the main error messages we can see here are `Fatal error.` and `SystemExit: 1`. I think @sagark had mentioned that he knew what was going wrong here and how to fix it.

sagark, 2018-10-08:
note: wrap https://github.com/firesim/firesim/blob/master/deploy/runtools/runtime_config.py#L109 in a try/catch, always write to log.

apaj, 2018-12-02:
Hi, I am facing a similar issue (only the line numbers are different):

```Running: infrasetup

Building FPGA software driver for FireSimNoNIC-FireSimRocketChipQuadCoreConfig-FireSimDDR3FRFCFSLLC4MBConfig
Fatal error.
Traceback (most recent call last):
  File "/home/centos/i-firesim/deploy/firesim", line 302, in <module>
    main(args)
  File "/home/centos/i-firesim/deploy/firesim", line 251, in main
    globals()[args.task](simconf)
  File "/home/centos/i-firesim/deploy/firesim", line 68, in infrasetup
    runtime_conf.infrasetup()
  File "/home/centos/i-firesim/deploy/runtools/runtime_config.py", line 240, in infrasetup
    self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
  File "/home/centos/i-firesim/deploy/runtools/firesim_topology_with_passes.py", line 293, in infrasetup_passes
    self.pass_build_required_drivers()
  File "/home/centos/i-firesim/deploy/runtools/firesim_topology_with_passes.py", line 276, in pass_build_required_drivers
    server.get_server_hardware_config().build_fpga_driver()
  File "/home/centos/i-firesim/deploy/runtools/runtime_config.py", line 110, in build_fpga_driver
    localcap = local("""make DESIGN={} TARGET_CONFIG={} PLATFORM_CONFIG={} f1""".format(design, target_config, platform_config), capture=True)
  File "/usr/lib64/python2.7/site-packages/fabric/operations.py", line 1239, in local
    error(message=msg, stdout=out, stderr=err)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 358, in error
    return func(message)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 62, in abort
    raise e
SystemExit: 1
The full log of this run is:
/home/centos/i-firesim/deploy/logs/2018-12-02--13-59-02-infrasetup-DC5H02MIEJYT8UFT.log
```

sagark, 2018-12-02:
Hi, you can run 

    make DESIGN=FireSimNoNIC TARGET_CONFIG=FireSimRocketChipQuadCoreConfig PLATFORM_CONFIG=FireSimDDR3FRFCFSLLC4MBConfig f1

in the `firesim/sim/` directory to see the actual error that happened while building the fpga driver.

I will push a commit shortly that will give this advice automatically when the manager encounters this problem (and log the error directly in the logfile).

sagark, 2018-12-04:
You can cherry-pick this commit if you want the more useful error message, which will automatically print out what I wrote in my previous message:
ae807e80e168dad8e380a1aa3a43310b9001b549

PR #192 (2018-12-03): Manager improvements
- Resolves #69, catch build error on infrasetup and log it to file + advise the user to run make command manually
- Resolves #100
closed 2019-02-25 by sagark