firesim/firesim #84: Running spec2017 does not work

pranith, 2018-09-01:
I tried the commands as listed in the guide. I trimmed the benchmark list to just run mcf. When I run the benchmark I get the following error in the output:

centos@ip-xx.ec2.internal:~/firesim/deploy/results-workload$ cat 2018-08-31--23-56-42-spec17-intspeed/605.mcf_s/output/605.mcf_s_.err 
./run.sh: line 4: ./: Is a directory
centos@ip-xx.ec2.internal:~/firesim/deploy/results-workload$ cat 2018-08-31--23-56-42-spec17-intspeed/605.mcf_s/output/605.mcf_s_.out 
Running: ./ inp.in 

Looks like the run.sh script is not picking up the benchmark name properly?

davidbiancolin, 2018-09-01:
Does the uartlog and generated init script make sense?

pranith, 2018-09-01:
The generated run script is not correctly populated with the binary name.

uartlog:
...
mount: mounting sysfs on /sys failed: No such device
Starting logging: OK
Starting mdev...
mdev: /sys/dev: No such file or directory
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... done.
Starting network: ip: SIOCGIFFLAGS: No such device
ip: can't find device 'eth0'
FAIL
Starting dropbear sshd: OK
Starting speed 605.mcf_s run with 1 threads
[    0.340000] reboot: Power down
Power off
time elapsed: 41.0 s, simulation speed = 60.54 MHz
*** PASSED *** after 2479894348 cycles
Runs 2479894348 cycles
[PASS] FireBoomNoNIC Test
SEED: 1535761358

Script done on Sat 01 Sep 2018 12:23:45 AM UTC
```

~/firesim/deploy/workloads/Speckle/build/overlay/intspeed/605.mcf_s$ cat run.sh 
#!/bin/bash
#This script was generated by Speckle gen_binaries.sh
echo 'Running: ./ inp.in '
./ inp.in 
centos@ip-192-168-5-91.ec2.internal:~/firesim/deploy/workloads/Speckle/build/overlay/intspeed/605.mcf_s$ cat run_workload0.sh 
#!/bin/bash
echo 'Running: ./ inp.in '
./ inp.in 

```

davidbiancolin, 2018-09-01:
Indeed, it is not. Where there any build errors during Speckle?

pranith, 2018-09-01:
No, the binaries are compiled properly. It is just the missing runscript.

davidbiancolin, 2018-09-01:
Relevant lines are around here: https://github.com/ccelio/Speckle/blob/firesim-2017/gen_binaries.sh#L158

This works for us; perhaps, something platform-specific is of? If the binaries are copied into the overlay correctly, i'd try to see what's going on in that script on your host.

pranith, 2018-09-01:
OK, I will retry a clean build of the benchmarks and see if it fixes it. Thanks!

closed 2018-09-01 by pranith