firesim/firesim #85: FireBoom config throws SIGILL for mcf from Spec2017

pranith, 2018-09-01:
Were the spec17 benchmarks tested for FireBoom? I am getting a SIGILL error while running mcf.

```
Starting speed 605.mcf_s run with 1 threads
[    0.336000] mcf_s_base.risc[96]: unhandled signal 4 code 0x1 at 0x00000000000101cc in mcf_s_base.riscv-64[10000+84000]
[    0.336000] CPU: 0 PID: 96 Comm: mcf_s_base.risc Not tainted 4.15.0-rc6-31580-g9c3074b5c2cd #1
[    0.336000] sepc: 00000000000101cc ra : 000000000001479c sp : 0000003fffe60c20
[    0.336000]  gp : 0000000000097100 tp : 0000000000099700 t0 : 0000000000000000
[    0.336000]  t1 : 2f2f2f2f2f2f2f2f t2 : 000000000000000c s0 : 0000000000014c04
[    0.336000]  s1 : 0000000000014cdc a0 : 0000000000000002 a1 : 0000003fffe60d68
[    0.336000]  a2 : 0000003fffe60d80 a3 : 0000000000000000 a4 : 0000003fffe60c48
[    0.336000]  a5 : 00000000000101cc a6 : 0000000000097320 a7 : 7efefefefefefeff
[    0.336000]  s2 : 0000000000000000 s3 : 00000000000d6008 s4 : 00000000000d6a88
[    0.336000]  s5 : 00000000000d4b88 s6 : 00000000000d4d68 s7 : 0000000000000000
[    0.336000]  s8 : 00000000000d6a48 s9 : 0000000000000000 s10: 000000000009c000
[    0.336000]  s11: 0000000000000000 t3 : 8101010101010100 t4 : 00000000000966b8
[    0.336000]  t5 : 00000000000956b8 t6 : 0000000000000000
[    0.336000] sstatus: 8000000200006020 sbadaddr: 00000000d5634785 scause: 0000000000000002
[    0.364000] reboot: Power down
```

pranith, 2018-09-01:
I guess this has something to do with BOOM not implementing compressed instructions?

jerryz123, 2018-09-01:
Yes this seems likely. You need to explicitly compile spec for rv64g, instead of rv64gc.

pranith, 2018-09-01:
@jerryz123 Do you knowhow to compile and check for rv64g in spec? Thanks!

jerryz123, 2018-09-01:
IIRC in `firesim/deploy/workloads/Speckle/riscv.cfg` you should change `rv64imafdc` to `rv64imafd`

pranith, 2018-09-01:
@jerryz123 thank you, that fixes the problem for me.

closed 2018-09-01 by pranith