firesim/firesim #88: Documentation regarding custom AFI/AGFI is unclear

pranith, 2018-09-05:
I tried following the documentation to create an agfi image for my modified BOOM core, but the instructions are not clear on what to do after successful building of the AGFI.

Section 4.3 ends by mentioning how to create a new fpga build, but stops short of mentioning how to deploy/run that custom built fpga and defers to the Advanced Docs section.

In the Advanced Docs section, we have instructions using make but no mention on how to customize and deploy the modified RTL.

davidbiancolin, 2018-09-05:
Hey Pranith, 

We'll update the docs, but if I understand you correctly, you've built your AGFI (aside -- now's a good time to check your design met timing) and are trying to deploy it F1 instances? 

The docs hint at this, but the names of things have changed somewhat. The bottom of section 4.3 has an email screenshot contains a snippet of what you should add to `deploy/config_hwdb.ini` (not `config_agfidb`).  If you didn't subscribe to SNS notifications that's ok -- you need to add to something of the form:

```
[<your-design-recipe-name-as-it-appears-in-config_build_recipes.ini]                                                                    
agfi=<your-designs-agfi>                                                                   
deploytripletoverride=None                                                                          
customruntimeconfig=None   
```

This is described in the [manager configuration docs](https://docs.fires.im/en/latest/Advanced-Usage/Manager/Manager-Configuration-Files.html#config-hwdb-ini).

Once you've added your AGFI to the hwdb, you can refer to it's name as you would any of the other example designs.

pranith, 2018-09-05:
Yes, I've built an agfi based on fireboom successfully. But I did not have SNS setup, so I did not get the email with the information. I found the agfi information in the results-build directory and added it to config_hwdb.ini as you mentioned. I was able to successfully deploy this fpga image. Thanks for the information.

Also, can you please point me to instructions on how to build a MegaBoom config fpga image?

sagark, 2018-09-05:
Glad that you got it working. We'd welcome PRs to the docs (as you've already done, thanks!) to improve them. Sometimes it's hard for us to see what's not clear since we built the infrastructure :)

To build a MegaBoom, you can create a config here based on BOOM's megaboom config:
https://github.com/firesim/firesim/blob/master/sim/src/main/scala/TargetConfigs.scala#L103

Once you do that, for early testing, you should go into firesim/sim/ and run

    make DESIGN=FireBoom TARGET_CONFIG=[the new config you made in TargetConfigs.scala] [make target you want to use, e.g. replace-rtl, emul]

pranith, 2018-09-06:
@sagark Thanks for the information. I was successfully able to create a custom target and run on the FPGA. I will try to add the missing information to the documentation.

Nanmu1996, 2022-03-07:
> I tried following the documentation to create an agfi image for my modified BOOM core, but the instructions are not clear on what to do after successful building of the AGFI.
> 
> Section 4.3 ends by mentioning how to create a new fpga build, but stops short of mentioning how to deploy/run that custom built fpga and defers to the Advanced Docs section.
> 
> In the Advanced Docs section, we have instructions using make but no mention on how to customize and deploy the modified RTL.
> 
> Can you please add instructions on modifying and running the custom FPGA AGFI?

Hello. I read your answer. There are a few questions. Can you help me please?  I want to create a new AGFI image, but I can't find a tutorial on how to do it. Sections 4.2 and 4.3 of the document are very brief. What is the input if you want to create a new AGFI? Do I need to submit a design document?