firesim/firesim #91: Unable to build integer only binaries with firesim toolchain

pranith, 2018-09-06:
I am not sure if this is the right place to report this. 

I am trying to build an integer only binary using -march=rv64i -mabi=lp64 using the firesim built toolchain. I am getting an error as follows:

riscv64-unknown-linux-gnu-gcc -c qht.c -std=c11 -ggdb3 -O2 -march=rv64ima -mabi=lp64 -fPIC
In file included from /home/centos/firesim/riscv-tools-install/sysroot/usr/include/features.h:447:0,
                 from /home/centos/firesim/riscv-tools-install/sysroot/usr/include/pthread.h:21,
                 from qht.h:4,
                 from qht.c:7:
/home/centos/firesim/riscv-tools-install/sysroot/usr/include/gnu/stubs.h:17:11: fatal error: gnu/stubs-lp64.h: No such file or directory
 # include <gnu/stubs-lp64.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.

a0u, 2018-09-07:
The default target for the toolchain is `rv64gc` and so the default ABI is `lp64d`.  You will need to rebuild riscv-gnu-toolchain after explicitly configuring using `--with-arch=rv64ima --with-abi=lp64` or the `--enable-multilib` option (which includes `rv64imac-lp64`) [1].

1. `git -C target-design/firechip submodule update --init riscv-tools`
1. Modify the riscv-gnu-toolchain configure options in [target-design/firechip/riscv-tools/build.sh](https://github.com/firesim/riscv-tools/blob/1859daea5e71809af62a6d4eb12d8f203c7efbc8/build.sh#L13).
1. Rerun `./build-setup.sh` in the firesim root without the `fast` option.

[1] https://github.com/riscv/riscv-gnu-toolchain/blob/master/README.md#installation-linux

pranith, 2018-09-07:
@a0u Thanks for the instructions. Is there any reason why firesim is not built with multilib already enabled?

a0u, 2018-09-07:
@pranith: The multilib combinations officially maintained in riscv-gnu-toolchain for glibc all include the C extension, which BOOM does not support yet.

Note that a multilib toolchain is only the first step in supporting the typical scenario of a Linux-based software stack - you still need to rebuild the userland with the correct arch/ABI for your target design.  Rather than deal with the complexity of arbitrary arch/ABI combinations, we simply opted to ship a single toolchain for `rv64g-lp64d`, which covers the common case for FireSim.  (I correct my earlier comment about the default target - `c` is omitted.)

Of course, we are open to documentation improvements that explain how to build and use a custom toolchain.

closed 2018-10-08 by sagark