firesim/firesim #187: Critical Warnings in Vivado Build

alonamid, 2018-11-29:
This issue might be more relevant to the aws-fpga repo.

Vivado logs output quite a few critical warnings.

Some of them include warnings regarding the SimpleNIC BRAMQueue URAM contraints, claiming that it can't find the relevant cells. However, the Queues are indeed mapped with the correct urams, and removing the contraints removes the urams. Unclear about the actual meaning of the warning:

```2018-11-29 11:54:10,133 [flush       ] [INFO ]  [192.168.2.248] out: WARNING: [Vivado 12-180] No cells matched 'firesim_top/top/SimpleNICWidget_0/incomingPCISdat/BRAMQueue/fq/ram_reg'. [/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/constraints/cl_synth_user.xdc:5]
2018-11-29 11:54:10,133 [flush       ] [INFO ]  [192.168.2.248] out: CRITICAL WARNING: [Common 17-55] 'set_property' expects at least one object. [/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/constraints/cl_synth_user.xdc:5]
2018-11-29 11:54:10,134 [flush       ] [INFO ]  [192.168.2.248] out: Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.
2018-11-29 11:54:10,134 [flush       ] [INFO ]  [192.168.2.248] out: WARNING: [Vivado 12-180] No cells matched 'firesim_top/top/SimpleNICWidget_0/outgoingPCISdat/BRAMQueue/fq/ram_reg'. [/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/constraints/cl_synth_user.xdc:6]
2018-11-29 11:54:10,134 [flush       ] [INFO ]  [192.168.2.248] out: CRITICAL WARNING: [Common 17-55] 'set_property' expects at least one object. [/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/constraints/cl_synth_user.xdc:6]
2018-11-29 11:54:10,134 [flush       ] [INFO ]  [192.168.2.248] out: Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.
```

davidbiancolin, 2018-11-30:
Yeah, aws-fpga is a better place for this.

alonamid, 2018-11-30:
Moved to aws-fpga

closed 2018-11-30 by alonamid