firesim/firesim #201: Makefile targets for individual MIDAS-level tests broken

dnshp, 2019-01-10:
Building a single test doesn't work as expected - there seems to be an issue with the Makefile rule. I entered the following:
```
make veryclean
make DESIGN=FireSimNoNIC
make DESIGN=FireSimNoNIC verilator-debug
make $(pwd)/output/f1/FireSimNoNIC-FireSimRocketChipConfig-FireSimConfig/rv64ui-p-simple.vpd
```
resulting in the following error
`make: *** No rule to make target '/home/centos/firesim/sim/output/f1/FireSimNoNIC-FireSimRocketChipConfig-FireSimConfig/rv64ui-p-simple.vpd'.  Stop.`
However, `make -j run-asm-tests` works as expected.

davidbiancolin, 2019-01-10:
What happens if you try: 
`make DESIGN=FireSimNoNIC $(pwd)/output/f1/FireSimNoNIC-FireSimRocketChipConfig-FireSimConfig/rv64ui-p-simple.vpd1`

dnshp, 2019-01-11:
That works correctly - is it possible to edit [this page](http://docs.fires.im/en/latest/Advanced-Usage/Debugging/RTL-Simulation.html#midas-level-simulation) for clarity?

sagark, 2019-01-12:
@dnshp can you PR a change to the dev branch to clarify? The file to edit is https://github.com/firesim/firesim/blob/dev/docs/Advanced-Usage/Debugging/RTL-Simulation.rst

PR #203 (2019-01-13): Added DESIGN to RTL simulation docs for clarity
Resolves #201.
closed 2019-01-13 by dnshp