firesim/firesim #205: Linux hangs while booting on a 3ClockDivConfig platform configuration

farzadfch, 2019-01-16:
Linux hangs at a random point near the end of boot on FireSimDDR3FRFCFSLLC4MB3ClockDivConfig platform configuration. Sometimes the simulation terminates and on the subsequent tries to run the simulation on the same F1 instance, the boot stops with a kernel panic message which appears to be different on each try.

The triplet and AGFI:
FireSimNoNIC-FireSimRocketChipQuadCoreConfig-FireSimDDR3FRFCFSLLC4MB3ClockDivConfig
agfi-0bea4cfabccaad045

One panic message:
```
[    0.024000] disk [generic-blkdev] of loaded; 3800800 sectors, 1 tags, 16 max request length
[    0.028000] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
[    0.028000] Oops [#1]
[    0.028000] CPU: 1 PID: 57 Comm: kthreadd Tainted: G        W        4.15.0-rc6-31587-gcae6324ee357 #1
[    0.028000] sepc: ffffffe00019baa8 ra : ffffffe000178468 sp : ffffffe3ef7f9a80
[    0.028000]  gp : ffffffe0003d6e00 tp : ffffffe3f00d7700 t0 : 0000000000000033
[    0.028000]  t1 : ffffffe00048c630 t2 : 0000000000000019 s0 : ffffffe3ef7f9ae0
[    0.028000]  s1 : ffffffe3ef7e0b90 a0 : ffffffe3ef7d8000 a1 : ffffffe3ef7e0b90
[    0.028000]  a2 : 0000000000000000 a3 : 0000000000000000 a4 : 0000000000000000
[    0.028000]  a5 : 0000000000000000 a6 : ffffffffffffffe0 a7 : 0000000000000000
[    0.028000]  s2 : 0000000000000000 s3 : ffffffe3ef7e0b90 s4 : 0000000000000100
[    0.028000]  s5 : ffffffe3effb25b8 s6 : 0000000000000001 s7 : 0000000000000002
[    0.028000]  s8 : ffffffe0003d3ac0 s9 : ffffffe3effb6880 s10: 0000000000000000
[    0.028000]  s11: 0000000000000001 t3 : 0000000000000014 t4 : ffffffe3ef7f9b00
[    0.028000]  t5 : 0000000000000068 t6 : 000000000000004c
[    0.028000] sstatus: 0000000200000100 sbadaddr: 0000000000000008 scause: 000000000000000d
[    0.028000] ---[ end trace 4071dd6ba6a494df ]---
[    0.028000] Kernel panic - not syncing: Fatal exception in interrupt
```

davidbiancolin, 2019-01-20:
So i'm pretty confident this a memory-model-side bug, and the clock domain crossings are fine.

davidbiancolin, 2019-02-02:
Hey @farzadfch ,

You don't happen to still have the complete uartlog & runworkload.log do you?  Could you email them to me at biancolin@eecs.<you know the rest>

farzadfch, 2019-02-03:
Hi David,

Please check your email.

farzadfch, 2019-03-21:
Hi David,

Have you had the chance to work on this?

davidbiancolin, 2019-03-25:
Hi Farzad, 

I haven't had the bandwidth to look into this because i've been overloaded with teaching this semester (i'll be far more free in May).  If this feature is important to you in the short term, i'd highly encourage you to try taking a pass at this using some of FireSim's debugging features. I'd be more than happy to walk you through the process of debugging in this issue.

There was a non-determinism bug in the RV channels i fixed earlier this semester that might manifest as a functional problem in the CDC channels, but given that applying the fix will be non-trivial. So one place to start would be to try running the FASEDTests AXI4Fuzzer driving the same memory system as your target to see if you can get it to fail when using a by-3 clock division. 

If you can i should be able to reproduce it and debug it in RTL sim very quickly.