firesim/firesim #226: error running firesim managerinit

eahmedsalman, 2019-02-01:
i got this error when i ran `firesim managerinit` on AWS
`Creating initial config files from examples.
Fatal error.
Traceback (most recent call last):
  File "/home/centos/i-firesim/deploy/firesim", line 302, in <module>
    main(args)
  File "/home/centos/i-firesim/deploy/firesim", line 235, in main
    managerinit()
  File "/home/centos/i-firesim/deploy/firesim", line 55, in managerinit
    m = local("""sed -i 's/AWSUSERNAME/{}/g' config_{}.ini""".format(get_aws_userid(), conf_file), capture=True)
  File "/home/centos/i-firesim/deploy/awstools/awstools.py", line 35, in get_aws_userid
    return client.get_user()['User']['UserId'].lower()
  File "/usr/lib/python2.7/site-packages/botocore/client.py", line 314, in _api_call
    return self._make_api_call(operation_name, kwargs)
  File "/usr/lib/python2.7/site-packages/botocore/client.py", line 612, in _make_api_call
    raise error_class(parsed_response, operation_name)
ClientError: An error occurred (AccessDenied) when calling the GetUser operation: User: arn:aws:iam::775658146153:user/ahmedsalman is not authorized to perform: iam:GetUser on resource: user ahmedsalman
The full log of this run is:
/home/centos/i-firesim/deploy/logs/2019-01-31--23-22-12-managerinit-SYG7P5K763ZNZBAE.log`

does any body know why? i just followed the docs

davidbiancolin, 2019-02-01:
Your IAM user doesn't have permission to look up  it's user ID. We use the ID to give a unique name to the S3 bucket we push FPGA build checkpoints to. We'll fix this, but for now you can change:

`("""sed -i 's/AWSUSERNAME/{}/g' config_{}.ini""".format(get_aws_userid(), conf_file), capture=True)`
to 
`("""sed -i 's/AWSUSERNAME/{}/g' config_{}.ini""".format(<your_desired_string_here>, conf_file), capture=True)`

on line 55 of deploy/firesim (in def managerinit())

davidbiancolin, 2019-02-01:
If you administer that IAM user, you could also give yourself the requisite permissions.

sagark, 2019-02-01:
Hi @eahmedsalman, can you tell us if running:

    curl -s http://169.254.169.254/latest/dynamic/instance-identity/document | grep -oP '(?<="accountId" : ")[^"]*(?=")'

On your instance prints out your user id? From our testing this seems to be a way to check without having the required IAM permissions set. If this works for you, I'll push a fix to the manager to avoid this issue.

eahmedsalman, 2019-02-01:
@sagark your command output is 
775658146153

eahmedsalman, 2019-02-02:
@sagark  I still  get the error `If you are a new user, supply your email address [abc@xyz.abc] for email notifications (leave blank if you do not want email notifications): e.ahmedsalman@gmail.com
Fatal error.
Traceback (most recent call last):
  File "/home/centos/i-firesim/deploy/firesim", line 302, in <module>
    main(args)
  File "/home/centos/i-firesim/deploy/firesim", line 235, in main
    managerinit()
  File "/home/centos/i-firesim/deploy/firesim", line 61, in managerinit
    subscribe_to_firesim_topic(useremail)
  File "/home/centos/i-firesim/deploy/awstools/awstools.py", line 284, in subscribe_to_firesim_topic
    Endpoint=email
  File "/usr/lib/python2.7/site-packages/botocore/client.py", line 314, in _api_call
    return self._make_api_call(operation_name, kwargs)
  File "/usr/lib/python2.7/site-packages/botocore/client.py", line 612, in _make_api_call
    raise error_class(parsed_response, operation_name)
AuthorizationErrorException: An error occurred (AuthorizationError) when calling the Subscribe operation: User: arn:aws:iam::775658146153:user/ahmedsalman is not authorized to perform: SNS:Subscribe on resource: arn:aws:sns:us-east-1:775658146153:FireSim
The full log of this run is:
/home/centos/i-firesim/deploy/logs/2019-02-02--01-05-49-managerinit-7Q7647AH8UN7QB1S.log
`
but when I tried it again without entering an email it didn't produce error 
`If you are a new user, supply your email address [abc@xyz.abc] for email notifications (leave blank if you do not want email notifications): 
You did not supply an email address. No notifications will be sent.
FireSim Manager setup completed.
The full log of this run is:
/home/centos/i-firesim/deploy/logs/2019-02-02--01-08-01-managerinit-BVJ3WQVXPEQ5856S.log
`