firesim/firesim #228: Asserts firing varying times instead of deterministically

jerryz123, 2019-02-01:
**Issue:** 
When debugging a buggy design with assertion synthesis and AutoILA, I observe an assert firing nondeterministically. 

**What I did**
I have a custom config `firemediumboom-singlecore-no-nic-lbp-asserts`  with a bunch of FpgaDebugAnnotations. I am on https://github.com/firesim/firesim/commit/448b4a6bcb0c85eafe2db312b5abb3e7c9b777eb.
I run the fedora-uniform workload, and observe that an assert is firing at varying times.

jerryz123, 2019-02-01:
The assertion firing is the target bug I'm trying to resolve. 

I need the assertion to fire at the same cycle between runs to debug this.

jerryz123, 2019-02-01:
Sorry,
Even without connecting any ILA triggers (but on a build with ILA wired up), I observe the target assertion firing on different target-cycles.

I am just running `firesim infrasetup && firesim runworkload` repeatedly, and the last lines in the uartlog display different cycle-counts. 

```
Assertion failed: Pipeline has hung.
    at core.scala:1190 assert (!(idle_cycles.value(13)), \"Pipeline has hung.\")
 at cycle: 2341074374

*** FAILED *** (code = 300) after 2341074390 cycles
time elapsed: 36.6 s, simulation speed = 63.94 MHz
FPGA-Cycles-to-Model-Cycles Ratio (FMR): 1.44
Runs 2341074390 cycles
[FAIL] FireBoomNoNIC Test
SEED: 1549048695
 at cycle 4294967295
```
vs
```
Assertion failed: Pipeline has hung.
    at core.scala:1190 assert (!(idle_cycles.value(13)), \"Pipeline has hung.\")
 at cycle: 4581695121

*** FAILED *** (code = 300) after 4581695137 cycles
time elapsed: 66.1 s, simulation speed = 69.36 MHz
FPGA-Cycles-to-Model-Cycles Ratio (FMR): 1.33
Runs 4581695137 cycles
[FAIL] FireBoomNoNIC Test
SEED: 1549048406
 at cycle 8589934590
```

System bus width is 64b.

jerryz123, 2019-02-02:
This seems to be an issue regardless of whether a ramdisk or the block device is used. The timestamp in `*** FAILED *** (code = 300) after 4581695137 cycles` varies in both cases.

I notice that SEED is different between runs. Could this be an issue?

davidbiancolin, 2019-02-02:
The seed is unused. I'd try synthesizing the commit log, that might be a good way to hunt down the point of divergence.

jerryz123, 2019-02-02:
Synthesized prints and assert should still print out to uartlog without the target WithUART mixin?

davidbiancolin, 2019-02-02:
yeah. they are handled separately.

davidbiancolin, 2020-04-02:
Resolved in 1.9.

closed 2020-04-02 by davidbiancolin