firesim/firesim #233: fedora booting stuck kernel panic message

eahmedsalman, 2019-02-05:
I got this got stuck here when trying to boot fedora
`[    0.172000] CPU: 0 PID: 1 Comm: systemd Not tainted 4.15.0-rc6-31587-gcae6324ee357 #2 
[    0.172000] sepc: ffffffe000175d78 ra : ffffffe000175f1c sp : ffffffe3efe4fb70 
[    0.172000]  gp : ffffffe00054da80 tp : ffffffe3efe60000 t0 : 0000000000000180 
[    0.172000]  t1 : 0000000000000000 t2 : 0000000000000000 s0 : ffffffe3efe4fba0 
[    0.172000]  s1 : ffffffe3efa8e440 a0 : fffffffffffffff6 a1 : ffffffe3efa8e440 
[    0.172000]  a2 : 0000000000000002 a3 : ffffffe000521600 a4 : 0000000000000000 
[    0.172000]  a5 : 0000000000000000 a6 : ffffffe0005fac18 a7 : 0000000000040000 
[    0.172000]  s2 : 0000000000000001 s3 : ffffffe3efe30000 s4 : fffffffffffff000 
[    0.172000]  s5 : ffffffe3ef26a021 s6 : ffffffe3efe4fd08 s7 : fffffffffffffff6 
[    0.172000]  s8 : 000000000000002f s9 : 0000000000000002 s10: 0000000000000000 
[    0.172000]  s11: 0000000000000000 t3 : 0000000000000000 t4 : 0000000000000000 
[    0.172000]  t5 : 000000000000006d t6 : ffffffe3ef93eb78 
[    0.172000] sstatus: 0000000200000120 sbadaddr: 0000000000000068 scause: 000000000000000d 
[    0.176000] ---[ end trace ad16cd37b4a2ef51 ]--- 
[    0.176000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b 
[    0.176000]
`

sagark, 2019-02-05:
Hi @eahmedsalman could you provide the full boot log (including the output before the error)

eahmedsalman, 2019-02-06:
`Script started, file is uartlog 
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000 
Using edma queue: /dev/edma0_queue_0 
UART0 is here (stdin/stdout). 
random min: 0x0, random max: 0xffffffffffffffff 
Zeroing out FPGA DRAM. This will take a few seconds... 
Commencing simulation. 
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000 
[    0.000000] Linux version 4.15.0-rc6-31587-gcae6324ee357 (centos@ip-192-168-4-196.ec2.internal) (gcc version 7.2.0 (GCC)) #2 SMP Tue Feb 5 15:45:34 UTC 2019 
[    0.000000] bootconsole [early0] enabled 
[    0.000000] Zone ranges: 
[    0.000000]   DMA      [mem 0x0000000080200000-0x00000000ffffffff] 
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000047fffffff] 
[    0.000000] Movable zone start for each node 
[    0.000000] Early memory node ranges 
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000047fffffff] 
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000047fffffff] 
[    0.000000] elf_hwcap is 0x112d 
[    0.000000] percpu: Embedded 16 pages/cpu @        (ptrval) s25624 r8192 d31720 u65536 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 4136455 
[    0.000000] Kernel command line: console=hvc0 root=/dev/generic-blkdev rw 
[    0.000000] Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes) 
[    0.000000] Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes) 
[    0.000000] Sorting __ex_table... 
[    0.000000] Memory: 16514888K/16775168K available (4134K kernel code, 208K rwdata, 922K rodata, 165K init, 801K bss, 260280K reserved, 0K cma-reserved) 
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1 
[    0.000000] Hierarchical RCU implementation. 
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1. 
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1 
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0 
[    0.000000] riscv,cpu_intc,0: 64l local interrupts mapped 
[    0.000000] riscv,plic0,c000000: mapped 2 interrupts to 1/2 handlers 
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns 
[    0.000000] console [hvc0] enabled 
[    0.000000] console [hvc0] enabled 
[    0.000000] bootconsole [early0] disabled 
[    0.000000] bootconsole [early0] disabled 
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=4000) 
[    0.000000] pid_max: default: 32768 minimum: 301 
[    0.000000] Mount-cache hash table entries: 32768 (order: 6, 262144 bytes) 
[    0.000000] Mountpoint-cache hash table entries: 32768 (order: 6, 262144 bytes) 
[    0.000000] Hierarchical SRCU implementation. 
[    0.000000] smp: Bringing up secondary CPUs ... 
[    0.000000] smp: Brought up 1 node, 1 CPU 
[    0.000000] devtmpfs: initialized 
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns 
[    0.000000] futex hash table entries: 256 (order: 2, 16384 bytes) 
[    0.004000] random: get_random_u32 called from bucket_table_alloc+0x134/0x3c0 with crng_init=0 
[    0.004000] NET: Registered protocol family 16 
[    0.004000] vgaarb: loaded 
[    0.004000] clocksource: Switched to clocksource riscv_clocksource 
[    0.008000] NET: Registered protocol family 2 
[    0.008000] TCP established hash table entries: 131072 (order: 8, 1048576 bytes) 
[    0.008000] TCP bind hash table entries: 65536 (order: 8, 1048576 bytes) 
[    0.008000] TCP: Hash tables configured (established 131072 bind 65536) 
[    0.008000] UDP hash table entries: 8192 (order: 6, 262144 bytes) 
[    0.008000] UDP-Lite hash table entries: 8192 (order: 6, 262144 bytes) 
[    0.008000] NET: Registered protocol family 1 
[    0.008000] workingset: timestamp_bits=62 max_order=22 bucket_order=0 
[    0.016000] random: fast init done 
[    0.016000] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254) 
[    0.016000] io scheduler noop registered 
[    0.016000] io scheduler cfq registered (default) 
[    0.016000] io scheduler mq-deadline registered 
[    0.016000] io scheduler kyber registered 
[    0.052000] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled 
[    0.060000] loop: module loaded 
[    0.060000] disk [generic-blkdev] of loaded; 12584960 sectors, 1 tags, 16 max request length 
[    0.060000] NET: Registered protocol family 10 
[    0.060000] Segment Routing with IPv6 
[... 5 lines cut ...]
[    0.064000] devtmpfs: mounted 
[    0.064000] Freeing unused kernel memory: 164K 
[    0.064000] This architecture does not have kernel memory protection. 
[    0.164000] systemd[1]: System time before build time, advancing clock. 
[    0.172000] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000068 
[    0.172000] Oops [#1] 
[    0.172000] Modules linked in: 
[    0.172000] CPU: 0 PID: 1 Comm: systemd Not tainted 4.15.0-rc6-31587-gcae6324ee357 #2 
[    0.172000] sepc: ffffffe000175d78 ra : ffffffe000175f1c sp : ffffffe3efe4fb70 
[    0.172000]  gp : ffffffe00054da80 tp : ffffffe3efe60000 t0 : 0000000000000180 
[    0.172000]  t1 : 0000000000000000 t2 : 0000000000000000 s0 : ffffffe3efe4fba0 
[    0.172000]  s1 : ffffffe3efa8e440 a0 : fffffffffffffff6 a1 : ffffffe3efa8e440 
[    0.172000]  a2 : 0000000000000002 a3 : ffffffe000521600 a4 : 0000000000000000 
[... cut ...]
[    0.172000]  s5 : ffffffe3ef26a021 s6 : ffffffe3efe4fd08 s7 : fffffffffffffff6 
[    0.172000]  s8 : 000000000000002f s9 : 0000000000000002 s10: 0000000000000000 
[    0.172000]  s11: 0000000000000000 t3 : 0000000000000000 t4 : 0000000000000000 
[    0.172000]  t5 : 000000000000006d t6 : ffffffe3ef93eb78 
[    0.172000] sstatus: 0000000200000120 sbadaddr: 0000000000000068 scause: 000000000000000d 
[    0.176000] ---[ end trace ad16cd37b4a2ef51 ]--- 
[    0.176000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b 
[    0.176000] 

`

eahmedsalman, 2019-02-06:
I want to point that I am using our core which is rocketchip with some modifications

sagark, 2019-02-07:
Since you have a custom target design, you'll need to use the debugging tools in FireSim to debug it:
http://docs.fires.im/en/latest/Advanced-Usage/Debugging/index.html

You can also confirm that your software stack boots on a vanilla copy of Rocket Chip

sagark, 2019-02-19:
Closing this since it is not an issue with an included target design.

closed 2019-02-19 by sagark