firesim/firesim #243: Cannot Execute Cross-compiler

redpanda3, 2019-02-23:
When I run 
./sw-manager.py -c br-disk.json build

in targeting software.

`Cannot execute cross-compiler '/home/centos/src/project_data/firesim/riscv-tools-install/bin/riscv64-unknown-linux-gnu-gcc'
make[2]: *** [/home/centos/src/project_data/firesim-nvdla/sw/firesim-software/br/buildroot/output/build/toolchain-external-custom/.stamp_configured] Error 1
make[1]: *** [_all] Error 2
make[1]: Leaving directory `/home/centos/src/project_data/firesim-nvdla/sw/firesim-software/br/buildroot'
`

sagark, 2019-02-23:
Can you tell us what version of the FPGA dev AMI you're on?

Also, it looks like you're using firesim-nvdla, so roping in @farzadfch here, who maintains that, since I don't know the details of the toolchain there.

farzadfch, 2019-02-23:
I am guessing that you have not sourced `sourceme-f1-manager.sh` in `firesim-nvdla`. The make error says it is looking for gcc in `firesim` directory.

As a side note, the `nvdla` branch does not make any changes to the toolchain.

farzadfch, 2019-02-26:
Hey @redpanda3,
Were you able to successfully run this? 
I opened the "Issues" tab on [FireSim-NVDLA](https://github.com/CSL-KU/firesim-nvdla). Feel free to ask any questions there, if you face an issue.

redpanda3, 2019-02-26:
> Hey @redpanda3,
> Were you able to successfully run this?
> I opened the "Issues" tab on [FireSim-NVDLA](https://github.com/CSL-KU/firesim-nvdla). Feel free to ask a question there, if you face an issue.

Haven't tried yet since last time. But I'm sure I sourced the sourceme-f1-manager.sh last time (when I open an issue). Maybe it's the 1.5.0 and 1.4.0 AMI version compatibility problem.

farzadfch, 2019-02-26:
I tried it on AMI 1.5.0 and it works fine.
The problem is most probbably with PATH not being set correctly. In the error you copied, it is looking for gcc in `firesim` (but your toolchain is located in `firesim-nvdla`) . When you source sourceme-f1-manager.sh, it calls env.sh. That's where PATH is being set.  Open env.sh and make sure PATH is beging set correclty. For you, it should look like this:

```
export RISCV=/home/centos/src/project_data/firesim-nvdla/riscv-tools-install
export PATH=/home/centos/src/project_data/firesim-nvdla/riscv-tools-install/bin:/home/centos/src/project_data/firesim-nvdla/:$PATH
export LD_LIBRARY_PATH=/home/centos/src/project_data/firesim-nvdla/riscv-tools-install/lib
```

I fixed a bug that wasn't related to this. Make sure you do a git pull before trying again.

closed 2019-05-23 by davidbiancolin