firesim/firesim #270: object util is not a member of package firesim

lorenzoditucci, 2019-04-02:
Hi all, 

I am trying to build a Firesim FPGAImage as described in (https://docs.fires.im/en/latest/Building-a-FireSim-AFI.html), however when I type firesim buildafi I get multiple errors such as:

/home/centos/firesim/sim/src/main/scala/firesim/Generator.scala:19:16: object util is not a member of package firesim

Attached the log of the simulation

Is there something I am doing wrong? Thanks
[2019-04-02--18-08-03-buildafi-R73OEYW0V70ZVKX9.log](https://github.com/firesim/firesim/files/3035355/2019-04-02--18-08-03-buildafi-R73OEYW0V70ZVKX9.log)

davidbiancolin, 2019-04-03:
Thanks for providing the log. Is this a fresh clone off master? 

This file exists in your local repo right?  
https://github.com/firesim/firesim/blob/master/sim/src/main/scala/util/GeneratorUtils.scala

lorenzoditucci, 2019-04-03:
Thanks for the reply. Yes, the file exists in my local repo. However, with a fresh clone of master, everything worked correctly.
It is strange as I do not recall changing any file in that folder - I guess I must have changed something and broke it!

davidbiancolin, 2019-04-03:
Cool. You could try removing all of your SBT target directories if you wanted to do some more sleuthing.

Something like `find -L . -iname target -type d | xargs rm -r` from sim/ should clear away all that state.

sagark, 2019-04-10:
Marking this as resolved

closed 2019-04-10 by sagark