firesim/firesim #274: Build instance runs out of disk space

farzadfch, 2019-04-03:
I don't know if this happens for others or not, but for me, Amazon decides to launch the build instance with 75GB of disk space because the block device config is not currently specified in the launch script. Out of this 75GB, only 5GB is free to be used by the build job and when the design is too large and Vivado runs out of space (I'm synthesizing NVDLA), it crashes with a misleading error message. [Increasing the disk volume size](https://github.com/CSL-KU/firesim-nvdla/blob/7d524d50ec27506c21174f8b3aa25c837cf220ca/deploy/buildtools/buildconfig.py#L38) fixes the issue.

This took me quite a while to figure out so thought that it might be helpful to others to avoid such headaches when building large designs.

davidbiancolin, 2019-04-03:
Out of curiosity, what was the error message?

sagark, 2019-04-03:
This is the same issue @a0u had in #265 (which hasn't been addressed yet).

davidbiancolin, 2019-04-03:
I'd like to see if it's the same message i'm getting. 

@farzadfch I think your fix is perfect, so long as the volume isn't persistent -- could you PR it?

farzadfch, 2019-04-03:
For me it runs out of space when it trys to write the post-route checkpoint:
```
2019-03-29 15:51:44,678 [flush       ] [INFO ]  [192.168.3.83] out:     Completed: route_design (WNS=0.018)
2019-03-29 15:51:44,678 [flush       ] [INFO ]  [192.168.3.83] out:     ################################
2019-03-29 15:51:44,678 [flush       ] [INFO ]  [192.168.3.83] out:     Writing route_design checkpoint: /home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/checkpoints/19_03_29-063720.post_route_design.dcp [Fri Mar 29 15:51:44 2019]
2019-03-29 15:51:44,678 [flush       ] [INFO ]  [192.168.3.83] out:
2019-03-29 15:52:10,085 [flush       ] [INFO ]  [192.168.3.83] out: Writing placer database...
2019-03-29 15:52:10,085 [flush       ] [INFO ]  [192.168.3.83] out: INFO: [Timing 38-480] Writing timing data to binary archive.
2019-03-29 15:53:25,276 [flush       ] [INFO ]  [192.168.3.83] out: Writing XDEF routing.
2019-03-29 15:53:25,641 [flush       ] [INFO ]  [192.168.3.83] out: Writing XDEF routing logical nets.
2019-03-29 15:53:43,872 [flush       ] [INFO ]  [192.168.3.83] out: Writing XDEF routing special nets.
2019-03-29 15:53:48,712 [flush       ] [INFO ]  [192.168.3.83] out: Write XDEF Complete: Time (s): cpu = 00:03:04 ; elapsed = 00:01:39 . Memory (MB): peak = 25181.031 ; gain = 0.000 ; free physical = 4944 ; free virtual = 47088
2019-03-29 16:04:30,382 [flush       ] [INFO ]  [192.168.3.83] out: ERROR: [Common 17-141] Failed to write file content of top_sp.xn in zip archive.
2019-03-29 16:04:30,382 [flush       ] [INFO ]  [192.168.3.83] out:
2019-03-29 16:04:30,883 [flush       ] [INFO ]  [192.168.3.83] out: ERROR: [Common 17-69] Command failed: Failed to create design checkpoint
2019-03-29 16:04:31,284 [flush       ] [INFO ]  [192.168.3.83] out: Abnormal program termination (11)
2019-03-29 16:04:31,284 [flush       ] [INFO ]  [192.168.3.83] out: Please check '/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/hs_err_pid2731.log' for details
2019-03-29 16:04:32,887 [flush       ] [INFO ]  [192.168.3.83] out:
2019-03-29 16:04:32,976 [flush       ] [INFO ]  Fatal error: run() received nonzero return code 139 while executing!
2019-03-29 16:04:32,976 [flush       ] [INFO ]  Requested: ./aws_build_dcp_from_cl.sh -foreground
2019-03-29 16:04:32,976 [flush       ] [INFO ]  Executed: /bin/bash -l -c "cd /home/centos/firesim-build/platforms/f1/aws-fpga && source hdk_setup.sh && export CL_DIR=/home/centos/firesim-build/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim && cd \$CL_DIR/build/scripts/ && ./aws_build_dcp_from_cl.sh -foreground"
2019-03-29 16:04:32,976 [flush       ] [INFO ]  Aborting.
2019-03-29 16:04:32,998 [<module>    ] [ERROR]  Fatal error.
```
@davidbiancolin sure, I'll make the PR.

PR #275 (2019-04-08): Increases the root volume size of the build instance to 100GB.
Resolves #274 and #265
closed 2019-06-26 by davidbiancolin