firesim/firesim #284: "current state: pending" in buildafi

msyksphinz, 2019-04-20:
Hi, 
I tried to buildafi firesim-singlecore-no-nic-lbp using f8ee092. I followed https://docs.fires.im/en/latest/Building-a-FireSim-AFI.html.
Running Vivado seemed to be successful, but finally I got following messages again and again (over 100~1000 times?)

```
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
Current state: pending
```

Log message said, "aws ec2 descripbe-fpga-images" and get folliwing message.

```
centos@ip-192-168-0-93.ec2.internal:~/firesim$ aws ec2 describe-fpga-images --fpga-image-id afi-0589f3e8a4e6e6e88 | tee AGFI_INFO
{
    "FpgaImages": [
        {
            "UpdateTime": "2019-04-20T07:00:55.000Z",
            "Name": "firesim-singlecore-no-nic-lbp",
            "Tags": [],
            "FpgaImageGlobalId": "agfi-0cf57a5c220c2ace3",
            "Public": false,
            "State": {
                "Code": "pending"
            },
            "OwnerId": "405410199528",
            "FpgaImageId": "afi-0589f3e8a4e6e6e88",
            "CreateTime": "2019-04-20T07:00:55.000Z",
            "Description": "firesim-buildtriplet:FireSimNoNIC-FireSimRocketChipSingleCoreConfig-FireSimConfig160MHz,firesim-deploytriplet:FireSimNoNIC-FireSimRocketChipSingleCoreConfig-FireSimConfig160MHz,firesim-commit:7f20a19cecdc43cc4909d666307f08daa4d17d0a-dirty"
        }
    ]
}
```

Should I wait more or something wrong?

sagark, 2019-04-21:
Once you're in this step, your AGFI is in final AWS-backend processing. This usually takes anywhere from 30 minutes to 1 hour. If it's gone longer than 2 hours, I would be concerned.

Did it finish for you?

msyksphinz, 2019-05-01:
Thank you for your support. I confirmed it has done about 30 minutes.

closed 2019-05-01 by msyksphinz