firesim/firesim #285: FireBoom MIDAS-level simulation does not print instruction trace in .out

farzadfch, 2019-04-23:
FireBoom MIDAS-level simulation does not print instruction trace in .out?

farzadfch, 2019-04-25:
Could you give me a pointer to the code which enables/disables this?

davidbiancolin, 2019-04-30:
@abejgonzalez @jerryz123

jerryz123, 2019-04-30:
I'm not sure what the instruction trace from MIDAS-level simulation is coming from. If it is generated by Chisel printfs, then you need to change a flag in `boom/src/main/scala/common/consts.scala`.

https://github.com/riscv-boom/riscv-boom/blob/80240845e700a43a144e66c46bbbe4479e00a73a/src/main/scala/common/consts.scala#L30

davidbiancolin, 2019-05-23:
@farzadfch Is this resolved?

farzadfch, 2019-05-23:
I was looking for a log which shows what the core is doing cycle by cycle and prints the instruction with DASM (similar to the Rocket Core's log). COMMIT_LOG_PRINTF does not do that (as the name suggests). [DEBUG_PRINTF](https://github.com/riscv-boom/riscv-boom/blob/80240845e700a43a144e66c46bbbe4479e00a73a/src/main/scala/exu/core.scala#L1168) looks closer to what I am looking for. I am going to try that one as soon as I find a time slot.

farzadfch, 2019-05-28:
DEBUG_PRINTF does print the log that I am looking for either. So I think we can close this issue.

ccelio, 2019-05-28:
`COMMIT_LOG_PRINTF` prints a list of committed instructions; which is not cycle-by-cycle, but can be compared against an ISA simulator like spike (and recommended to pass it through `spike-dasm` to translate the `DASM(0xABCDABCD)` into instructions.)

`DEBUG_PRINTF` prints out the internal state of the pipeline (ROB, issue queues, load/store queues, etc.), which can be easier to digest than waveforms in some situations.

farzadfch, 2019-05-30:
Thanks @ccelio, COMMIT_LOG_PRINTF did not print DASM(inst) so [I added](https://github.com/farzadfch/riscv-boom/commit/d1ef882295cd6562945d5867dce1e65753cdc9ca) that plus the hart ID in the log.

farzadfch, 2019-07-12:
I will send a PR to fix this.

farzadfch, 2019-07-13:
I think I had the changes that enable commit log from `Target.scala` without modifying the BOOM code but unfortunately, I have lost those changes.
This issue is resolved for me and I close it.

closed 2019-07-13 by farzadfch