firesim/firesim #293: Partially routed net in MemModel

a0u, 2019-05-22:
Vivado occasionally fails to route a clock net in the `MemModel` that apparently has been split between two SLRs:
```
CRITICAL WARNING: [Route 35-535] Clock Net: WRAPPER_INST/CL/firesim_top/top/MemModel_0/ingress/wQueue/wantREF10_out is not completely routed.
CRITICAL WARNING: [Route 35-8] Design has 7 unrouted pins, that are still reachable.
```
This subsequently leads to a DRC violation when `write_bitstream` is run during AGFI creation.  The issue seems to be non-deterministic and can manifest even when slightly changing a design in a functionally unrelated manner.

Here is the `report_route_status` summary:
```
Nets with Routing Errors:
  WRAPPER_INST/CL/firesim_top/top/MemModel_0/ingress/wQueue/wantREF10_out
    Unrouted Pins:
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/rQueue/maybe_full_reg/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/rQueue/value_1_reg[0]/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/rQueue/value_1_reg[1]/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/wQueue/maybe_full_reg/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/wQueue/value_1_reg[0]/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/wQueue/value_1_reg[1]/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/wQueue/value_reg[0]/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/wQueue/value_reg[1]/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/xactionRelease/currentRead/_T_37_reg/R
      WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/xactionRelease/currentWrite/_T_37_reg/R
```

The Vivado log contains further details about each unroutable connection:
```
Unroutable connection Types: 
----------------------------
Checking all reachable nodes within 5 hops of driver and load 

Unroute Type 1 : Site pin does not reach interconnect fabric

        Type 1 BUFGCE.CLK_OUT->SLICEL.SRST1
        -----Num Open nets: 3
        -----Representative Net: Net[66977] WRAPPER_INST/CL/firesim_top/top/MemModel_0/ingress/wQueue/wantREF10_out
        -----BUFGCE_X1Y273/CLK_OUT -> SLICE_X89Y559/SRST1
        -----Driver Term: WRAPPER_INST/CL/firesim_top/top/MemModel_0/ingress/wQueue/value_1[1]_i_1__169_bufg_place/O Load Term [280830]: WRAPPER_INST/CL/firesim_top/top/MemModel_0/model/backend/rQueue/maybe_full_reg/R
        Driver Pin does not reach Interconnect fabric within 5 hops.
        Pins Reached within 5 hops from Driver
        MMCM_X1Y11/CLKIN1 Net on Pin: 
        MMCM_X1Y11/CLKIN2 Net on Pin: 
        PLL_X1Y22/CLKIN Net on Pin: WRAPPER_INST/CL/SH_DDR/ddr_cores.DDR4_0/inst/u_ddr4_infrastructure/pll_gate_cnt_reg[4]_0  Net Driver: BUFGCE_X1Y308/CLK_OUT
        PLL_X1Y23/CLKIN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y278/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y279/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y280/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y281/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y282/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y283/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y275/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y284/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y285/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y286/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y287/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y288/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y289/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y290/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y291/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y276/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y292/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y294/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y295/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y296/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y297/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y298/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y299/CLK_IN Net on Pin: 
        GCLK_TEST_BUFE3_X130Y277/CLK_IN Net on Pin: 
        Type 2 BUFGCE.CLK_OUT->SLICEL.SRST2
...
```

I was unsuccessful in using a placement constraint to group the MemModel cells within the same SLR ([UG912](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug912-vivado-properties.pdf)), but judging from some warnings, I could be selecting the cells incorrectly.
```
set_property USER_SLR_ASSIGNMENT group_1 [get_cells -hierarchical -filter {NAME =~ *MemModel_0*}]
```

davidbiancolin, 2019-05-23:
This all started happening because i told Jan Gray that we _weren't_ having problems with multi-SLR designs.

farzadfch, 2020-02-26:
I am using an older version of firesim and I was wondering if a workaround has been found for this.

davidbiancolin, 2020-02-26:
Have you tried the SLR assignment constraint yourself? From Albert's comment it seems like it might not have been applied correctly.

How frequently do you see this issue?

farzadfch, 2020-02-28:
The net itself is not always in the memory model, though this is generally caused by a modification to it. For instance, I think this time the net is between a token channel and a tile's fronted (though I'm not sure. PFA [the log](https://github.com/firesim/firesim/files/4264929/log.txt)). So I don't know which part of the design I have to set the constraint on.

Do you know what is exactly causing this? I think the design is always split between two SLRs. So why this does not always manifest?

Answering your question, I think this has happened 3 out of 4 times that I have modified the mem model. This time I was trying to use LLC and LBP together.