firesim/firesim #298: error in 'firesim buildafi'

yswntht, 2019-06-14:
Hi all,

I am trying to build a Firesim FPGAImage. I see 'AttributeError: Raw' and build fails.

Here is the log. 
Am I missing something??
[2019-06-14--10-45-51-buildafi-L1PJN949JM8TYEH4.log](https://github.com/firesim/firesim/files/3290440/2019-06-14--10-45-51-buildafi-L1PJN949JM8TYEH4.log)

Thanks!

bkorpan, 2019-06-15:
I've had the same problem, it seems like the setup procedure for FireSim master is currently broken. For now, I'd recommend trying the dev branch and following the dev setup instructions: https://docs.fires.im/en/dev/Initial-Setup/index.html. The main difference is that you need to use the new FireSim AMI, which already includes the machine launch script.

davidbiancolin, 2019-06-18:
I'm looking into this now. I'd be careful about using dev, as we're not quite ready to roll out the FireSim AMI.

yswntht, 2019-06-19:
Yes! Also, how about using dev branch with FPGA AMI - would that work fine?

Thanks!

davidbiancolin, 2019-06-19:
Can you run `sudo pip2 install fabric` on your manager instance. That should resolve your problem.

yswntht, 2019-06-19:
well, I see it already present in the [flow](https://docs.fires.im/en/latest/Initial-Setup/Setting-up-your-Manager-Instance.html)

davidbiancolin, 2019-06-19:
That's just a quick and dirty way to get pip to fix the dependency on the cryptography package. If you want to be more precise you can try: `sudo pip2 install cryptography==2.5.0`

yswntht, 2019-06-19:
Thanks! issue resolved with your fix!

PR #299 (2019-06-19): Revert "downgrade package to remove deprecation warnings"
Resolves #298.

This breaks `buildafi`. We can deal with paramiko spewing garbage during `runworkload` for now.
closed 2019-06-19 by yswntht