firesim/firesim #305: Stall for bdev_rresp_stalled

armiasalib, 2019-06-29:


armiasalib, 2019-06-29:
During booting Fedora sometime the boot stuck after printing the following message:
`Started udev Kernel Device Manager.`

I can notice this stuck very rarely when I use one tracker for the DevBlk; however, after increasing the number of trackers to 8, I notice this issue more frequently (10%), and it appears 50% of the times when I use 32 cores.

I did some debug on this issue, and I see that `bdev_rresp_stalled` is asserted when this stuck happens. So I looked into the implementation of https://github.com/firesim/firesim/blob/1.5.0/sim/src/main/scala/endpoints/BlockDevWidget.scala I think there is a path that keep `readRespBeatsLeft=1` even after finishing all responses, that will make the system stall. This path happens when `done=1` and at the same time `readLatencyPipe.io.deq.valid=1`

I would suggest to set a default values for `readRespBeatsLeft` and `returnWrite` when `done || idle` condition is met. So the code will be something like:

```
    when (tFire) {
      when (done || idle) {
        readRespBeatsLeft := 0.U
        returnWrite := false.B

        // If a write-response is waiting, return it first
        when(writeLatencyPipe.io.deq.valid) {
          returnWrite := true.B
          writeLatencyPipe.io.deq.ready := true.B
        }.elsewhen(readLatencyPipe.io.deq.valid) {
          readRespBeatsLeft := readLatencyPipe.io.deq.bits * dataBeats.U
          readLatencyPipe.io.deq.ready := true.B
        }
      }.elsewhen(readRespBusy && target.resp.fire) {
        readRespBeatsLeft := readRespBeatsLeft - 1.U
      }
    }
```

Please let me know if I am missing something,

davidbiancolin, 2019-06-29:
Hey Armiaslib, 

Thanks for opening your first issue and proposing a solution! That's a good catch, and the fix looks good. How did you test it? It would be great if you could PR this change against master. Just run `make test` (or open the sbt shell, and run `testOnly firesim.firesim.*`) out of `sim/` first. I will regenerate the AGFIs and merge it into master once you're ready.

armiasalib, 2019-07-02:
We are developing a modified version of Rocket chip that we test on Firesim environment. I do testing on FPGA to make sure that Fedora can boot successfully and to run multiple benchmarks.

I will do PR this week after the recommended simulation verification.

armiasalib, 2019-07-04:
I did the pull request. Please let me know if you need any further information.

PR #308 (2019-07-08): Fix a Block Device Endpoint Bug 
Resolves #305. Adds regenerated AGFIs to #307, which this PR subsumes.

H/T @armiasalib.
closed 2019-07-08 by davidbiancolin