firesim/firesim #311: Using Vivado IP Cores in Target Design

sibanez12, 2019-07-11:
Is it possible to use a Vivado IP core within a FireSim target design?

After reading the section about [Restrictions on Target RTL](https://docs.fires.im/en/latest/Advanced-Usage/Generating-Different-Targets.html#restrictions-on-target-rtl) my impression is that this is not possible, but I would like to check if my understanding is correct.

Thanks!

davidbiancolin, 2019-07-11:
Hi Sibanez12,

It is not presently possible to do that in mainline FireSim, but there are ways to get this work. However, this question came up on the google groups [here](https://groups.google.com/forum/#!topic/firesim/0s-N0Qkwmlw). 

What kind of core is it?

sibanez12, 2019-07-12:
Thanks this is helpful :) I'd like to use a [Xilinx SDNet](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_2/ug1012-sdnet-packet-processor.pdf) generated packet-processing module in the NIC interface. The SDNet module exposes standard AXI4 Stream master/slave interfaces for the packet processing data-path and an AXI-Lite control interface.

If I'm understanding correctly, it should be fairly straight-forward to black box a Xilinx IP core and use that within the FireChip design, but it will require more effort to actually use this design in a FireSim simulation because the tools use a FIRRTL transformation to implement the simulation-stall logic?

Does this sound correct to you? I'm new to the RISC-V and Chisel world :)

davidbiancolin, 2019-07-12:
Yes that's correct -- you'd need to make changes to the FAME1 transform, as a few others have done. @farzadfch described how he did it [here](https://groups.google.com/forum/#!topic/firesim/_dFKGiQv4Es). You should be able to cherry-pick that commit. 

We're planning to have support for this style of BlackBox integration in the next release.

sibanez12, 2019-07-13:
Great, thanks a lot for the help!

closed 2019-07-13 by sibanez12