firesim/firesim #329: Copy results files fail

NathanTP, 2019-08-12:
Occasionally, the runfarm instance will fail to copy files back to the manager. Manually re-running the commands works. See [here](https://groups.google.com/forum/#!topic/firesim/GI29vu-93es) for the initial report.

Here is a trace provided by hzhuo2@wisc.edu:
2019-08-01 21:23:10,886 [instance_log] [INFO ]  [192.168.1.43] Slot 0 completed! copying results.
2019-08-01 21:23:10,886 [flush       ] [DEBUG]  [localhost] local: mkdir -p /home/centos/firesim/deploy/results-workload/2019-08-01--21-04-56-cus-fed29//cus-fed29-test/
2019-08-01 21:23:10,893 [copy_back_jo] [DEBUG]  [localhost] 
2019-08-01 21:23:10,893 [copy_back_jo] [DEBUG]  [localhost] 
2019-08-01 21:23:10,894 [flush       ] [DEBUG]  [192.168.1.43] run: sudo mkdir -p /home/centos/sim_slot_0/mountpoint
2019-08-01 21:23:11,063 [flush       ] [DEBUG]  [192.168.1.43] run: sudo mount /home/centos/sim_slot_0/cus-fed29-test.img /home/centos/sim_slot_0/mountpoint
2019-08-01 21:23:11,600 [flush       ] [DEBUG]  [192.168.1.43] run: sudo chmod -Rf 777 /home/centos/sim_slot_0/mountpoint
2019-08-01 21:23:13,551 [flush       ] [DEBUG]  Fatal error: run() received nonzero return code 1 while executing!
2019-08-01 21:23:13,551 [flush       ] [DEBUG]  Requested: sudo chmod -Rf 777 /home/centos/sim_slot_0/mountpoint
2019-08-01 21:23:13,551 [flush       ] [DEBUG]  Executed: /bin/bash -l -c "sudo chmod -Rf 777 /home/centos/sim_slot_0/mountpoint"
2019-08-01 21:23:13,551 [flush       ] [DEBUG]  Aborting.
2019-08-01 21:23:13,573 [<module>    ] [ERROR]  Fatal error.

NathanTP, 2019-08-14:
Reproduce in more detail: [runworkload.log](https://github.com/firesim/firesim/files/3499432/runworkload.log)

branch: dev
workload: fedora-base
HW: [config_runtime.txt](https://github.com/firesim/firesim/files/3499437/config_runtime.txt)
command: "firesim launchrunfarm && firesim infrasetup && firesim runworkload"

This was a freshly started instance, all I did in the simulation itself was ping between the two nodes and then "poweroff -f". Same behavior two runs in a row (second run didn't even ping, just boot and 'poweroff -f').

zhemao, 2019-08-14:
Annoying that it doesn't give you the specific return code. Maybe this is because the image hasn't been fully mounted before it tries to run `chmod`.

abejgonzalez, 2019-11-06:
The full error message (if you remove the `-f`) is 
`chmod: changing permissions of ‘/home/centos/sim_slot_0/mountpoint/etc/sysconfig/nfs’: Operation not permitted`

abejgonzalez, 2019-11-06:
Also this occurs when just doing a plain boot with fedora and shutting down

sagark, 2019-11-09:
One option for fixing this is probably to only chmod the files that the user wants to pull off, rather than the whole thing

NathanTP, 2019-11-11:
Ya, I think just copying and then chmoding is probably best (or use rsync with the --chown and --chmod options to do it in one step). This is what marshal does for local simulation.

a0u, 2020-01-14:
@abejgonzalez: Add the `-v` flag to the `chmod` command to see if there are more diagnostics about the error.  `/etc/sysconfig/nfs` should be a regular file, so it's not immediately clear why `sudo` would fail with EPERM here.  An extended file attribute might be involved.

There are a few potential options:
* `sudo chown -R centos /home/centos/sim_slot_0/mountpoint` (should be preferable to blindly setting the mode to 777)
* Run `chown`/`chmod` only on the files to be explicitly transferred (as suggested by @sagark).  Note the parent directories also  need permissions adjusted.  There isn't an inverse `-R` flag to do this, but it would be simple to iterate over the path components up to the mountpoint.
* Run `rsync` with `sudo` on the remote host (not desirable).

The `--chmod`/`--chown` options for `rsync` are not applicable since they affect only the destination files.

PR #456 (2020-01-20): Fix Copy Results Failure
Fixes #329 by removing the `immutable` attribute on the `nfs` file that was causing problems. This PR also changes the `chmod` to `chown` based on what @a0u recommends. 

This was tested in Chipyard with Fedora boot (`fedora-base.json`).
closed 2020-01-23 by abejgonzalez