firesim/firesim #401: FpgaDebug giving an error while building firesim-nvdla

sanjeev-murthy, 2019-10-18:
I am inserting FpgaDebug in a .scala file as directed in the example in the Firesim document.
`import midas.targetutils.FpgaDebug`
`class nvdla(blackboxName: String, hasSecondAXI: Boolean, dataWidthAXI: Int) extends BlackBox {`
`  val nvdla_core2dbb_aw_awvalid = Bool(OUTPUT)`
`  val nvdla_core2dbb_aw_awready = Bool(INPUT)`
`  FpgaDebug(nvdla_core2dbb_aw_awvalid,nvdla_core2dbb_aw_awready)`
`}`

This is the error message that comes on the prompt while running the make command in sim folder:
```mkdir -p /home/centos/firesim-nvdla/sim/generated-src/f1/FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig10MHz
sbt -J-Xmx16G ++2.12.4 \
"runMain firesim.firesim.FireSimGenerator midas generated-src/f1/FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig10MHz firesim.firesim FireSimNoNIC firesim.firesim FireSimRocketChipQuadCoreConfig_WithNVDLALarge firesim.firesim FireSimDDR3FRFCFSLLC4MBConfig10MHz"
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/centos/firesim-nvdla/sim/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Set current project to firesim (in build file:/home/centos/firesim-nvdla/sim/)
[info] Setting Scala version to 2.12.4 on 16 projects.
[info] Reapplying settings...
[info] Set current project to firesim (in build file:/home/centos/firesim-nvdla/sim/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running firesim.firesim.FireSimGenerator midas generated-src/f1/FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig10MHz firesim.firesim FireSimNoNIC firesim.firesim FireSimRocketChipQuadCoreConfig_WithNVDLALarge firesim.firesim FireSimDDR3FRFCFSLLC4MBConfig10MHz
[info] [0.002] Elaborating design...
[info] [0.053] Done elaborating.
Total FIRRTL Compile Time: 288.0 ms
[info] [0.000] Elaborating design...
Interrupt map (8 harts 3 interrupts):
  [1, 1] => uart_0
  [2, 2] => frontend
  [3, 3] => nvdla

[error] (run-main-0) scala.MatchError: null
[error] scala.MatchError: null
[error] 	at chisel3.core.BaseModule.assignCompatDir$1(Module.scala:274)
[error] 	at chisel3.core.BaseModule._bindIoInPlace(Module.scala:278)
[error] 	at Chisel.package$BlackBox._compatAutoWrapPorts(compatibility.scala:255)
[error] 	at chisel3.core.requireIsHardware$.apply(Binding.scala:26)
[error] 	at chisel3.core.dontTouch$.apply(Annotation.scala:84)
[error] 	at midas.targetutils.FpgaDebug$.$anonfun$apply$2(annotations.scala:28)
[error] 	at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:234)
[error] 	at scala.collection.IndexedSeqOptimized.foreach(IndexedSeqOptimized.scala:32)
[error] 	at scala.collection.IndexedSeqOptimized.foreach$(IndexedSeqOptimized.scala:29)
[error] 	at scala.collection.mutable.WrappedArray.foreach(WrappedArray.scala:38)
[error] 	at scala.collection.TraversableLike.map(TraversableLike.scala:234)
[error] 	at scala.collection.TraversableLike.map$(TraversableLike.scala:227)
[error] 	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[error] 	at midas.targetutils.FpgaDebug$.apply(annotations.scala:28)
[error] 	at nvidia.blocks.ip.dla.nvdla$$anon$1.<init>(nvdla.scala:96)
[error] 	at nvidia.blocks.ip.dla.nvdla.<init>(nvdla.scala:15)
[error] 	at nvidia.blocks.dla.NVDLA$$anon$1.$anonfun$u_nvdla$1(NVDLA.scala:87)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at nvidia.blocks.dla.NVDLA$$anon$1.<init>(NVDLA.scala:87)
[error] 	at nvidia.blocks.dla.NVDLA.module$lzycompute(NVDLA.scala:85)
[error] 	at nvidia.blocks.dla.NVDLA.module(NVDLA.scala:85)
[error] 	at nvidia.blocks.dla.NVDLA.module(NVDLA.scala:20)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:158)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:158)
[error] 	at scala.collection.immutable.List.flatMap(List.scala:335)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:156)
[... 43 lines cut ...]
[error] 	at sbt.Run.run0(Run.scala:87)
[error] 	at sbt.Run.execute$1(Run.scala:65)
[error] 	at sbt.Run.$anonfun$run$4(Run.scala:77)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error] 	at sbt.TrapExit$App.run(TrapExit.scala:252)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error] 	at sbt.Run$.executeTrapExit(Run.scala:124)
[error] 	at sbt.Run.run(Run.scala:77)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error] 	at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error] 	at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error] 	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] 	at scala.util.Try$.apply(Try.scala:209)
[error] 	at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 26 s, completed Oct 18, 2019 9:13:19 PM
make: *** [/home/centos/firesim-nvdla/sim/generated-src/f1/FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig10MHz/FireSimNoNIC-const.h] Error 1
```

Kindly tell what change is needed to get it to run. Thanks!

davidbiancolin, 2019-10-18:
Try annotating the nvdla black-box in the module in which it is instantiated, instead of in the definition of the black box.

JunaidCBNU, 2021-01-09:
Did u find the solution ?