firesim/firesim #410: fatal error at firesim launchrunfarm

CristinaZN, 2019-11-01:
hi, here is a problem.please help me
i create instance c4 and f1 both in us-east-1a

[centos@ip-192-168-0-187 deploy]$ firesim launchrunfarm
FireSim Manager. Docs: http://docs.fires.im
Running: launchrunfarm

An error occurred (VcpuLimitExceeded) when calling the RunInstances operation: You have requested more vCPU capacity than your current vCPU limit of 8 allows for the instance bucket that the specified instance type belongs to. Please visit http://aws.amazon.com/contact-us/ec2-request to request an adjustment to this limit.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (VcpuLimitExceeded) when calling the RunInstances operation: You have requested more vCPU capacity than your current vCPU limit of 8 allows for the instance bucket that the specified instance type belongs to. Please visit http://aws.amazon.com/contact-us/ec2-request to request an adjustment to this limit.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (VcpuLimitExceeded) when calling the RunInstances operation: You have requested more vCPU capacity than your current vCPU limit of 8 allows for the instance bucket that the specified instance type belongs to. Please visit http://aws.amazon.com/contact-us/ec2-request to request an adjustment to this limit.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (Unsupported) when calling the RunInstances operation: Your requested instance type (f1.2xlarge) is not supported in your requested Availability Zone (us-east-1f). Please retry your request by not specifying an Availability Zone or choosing us-east-1a, us-east-1b, us-east-1c, us-east-1d, us-east-1e.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (VcpuLimitExceeded) when calling the RunInstances operation: You have requested more vCPU capacity than your current vCPU limit of 8 allows for the instance bucket that the specified instance type belongs to. Please visit http://aws.amazon.com/contact-us/ec2-request to request an adjustment to this limit.
This probably means there was no more capacity in this availability zone. Try the next one.
An error occurred (VcpuLimitExceeded) when calling the RunInstances operation: You have requested more vCPU capacity than your current vCPU limit of 8 allows for the instance bucket that the specified instance type belongs to. Please visit http://aws.amazon.com/contact-us/ec2-request to request an adjustment to this limit.
This probably means there was no more capacity in this availability zone. Try the next one.
we tried all subnets, but there was insufficient capacity to launch your instances
only the following 0 instances were launched
[]
Waiting for instance boots: 0 f1.16xlarges
Waiting for instance boots: 0 f1.4xlarges
Waiting for instance boots: 0 m4.16xlarges
Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim-nvdla/deploy/firesim", line 307, in <module>
    main(args)
  File "/home/centos/firesim-nvdla/deploy/firesim", line 255, in main
    globals()[args.task](simconf)
  File "/home/centos/firesim-nvdla/deploy/firesim", line 139, in launchrunfarm
    runtime_conf.runfarm.launch_run_farm()
  File "/home/centos/firesim-nvdla/deploy/runtools/run_farm.py", line 244, in launch_run_farm
    wait_on_instance_launches(f1_2s, 'f1.2xlarges')
  File "/home/centos/firesim-nvdla/deploy/awstools/awstools.py", line 206, in wait_on_instance_launches
    rootLogger.info("Waiting for instance boots: " + str(len(instances)) + " " + message)
TypeError: object of type 'NoneType' has no len()
The full log of this run is:
/home/centos/firesim-nvdla/deploy/logs/2019-11-01--14-02-39-launchrunfarm-450NQXEVBSILY916.log

closed 2019-11-01 by CristinaZN

exciting678, 2019-12-23:
Hi, have you resolved the problem? I also meet the problem now. If you know, please inform me, thank you very much.

sagark, 2019-12-23:
You need to request a limit increase, as noted here in the firesim docs:

http://docs.fires.im/en/latest/Initial-Setup/First-time-AWS-User-Setup.html#requesting-limit-increases

exciting678, 2019-12-23:
Thank you. But there is no f1.2xlarge in US East (Northern Virginia) region, but in US East(GovCloud), so how to use f1.2xlarge for non-US citizens？

sagark, 2019-12-23:
You need to request an increase for "Running On-Demand All F instances". 

Thanks for pointing out the issue with the documentation, AWS recently changed how limits are configured so f1.2xlarge is no longer an option, I will update the docs to reflect this.

sagark, 2019-12-23:
@exciting678 can you give these new instructions a try? https://github.com/firesim/firesim/pull/448/files

exciting678, 2019-12-24:
What is the relationship between the region requested for an increase for "Running On-Demand All F instances" and "f1.2xlarge", do they have to be same?

exciting678, 2019-12-24:
Hi, I have try it and it works, thanks!