firesim/firesim #421: bump aws-fpga to use FPGA Dev AMI 1.7.0

sagark, 2019-11-07:
https://github.com/aws/aws-fpga/releases/tag/v1.4.11

timsnyder-siv, 2020-02-08:
Are there known issues when using the 1.7.0 AMI?

sagark, 2020-02-10:
I don't believe any of us have tried this, so I don't know of any
particular issues. However, the aws-fpga README suggests that it is not
compatible: https://github.com/aws/aws-fpga/#fpga-developer-ami
I would guess it will fail when the aws scripts check for vivado version.

sibanez12, 2020-05-28:
I would like to use Firesim with a Xilinx IP core that is only compatible with Vivado 2019.2+

This means I really want to use the FPGA Dev AMI 1.8.0.

Any idea how much headache I would run into if I try to merge [aws-fpga-firesim](https://github.com/firesim/aws-fpga-firesim) with the upstream [aws-fpga v1.4.14](https://github.com/aws/aws-fpga/tree/v1.4.14)?

Is there anything else I would need to do other than this merge and updating the `f1_ami_name` in [awstools.py](https://github.com/firesim/firesim/blob/master/deploy/awstools/awstools.py#L14)?

Thanks for the help!

sagark, 2020-06-11:
I don't think it should be too bad, what you've suggested should be it IIRC.