firesim/firesim #426: Followed the tutorial but linux won't start boot

Yangff, 2019-11-13:
I followed the docs on https://docs.fires.im/en/latest/Running-Simulations-Tutorial/Running-a-Single-Node-Simulation.html and try to get the `linux-uniform0` demo booting. 

From the `screen -r fsim0` all I get stopped after `Zeroing out FPGA DRAM. This will take a few seconds...`

```

Script started, file is uartlog
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000
Using xdma write queue: /dev/xdma0_h2c_0
Using xdma read queue: /dev/xdma0_c2h_0
UART0 is here (stdin/stdout).
command line for program 0. argc=46:
+permissive +mm_relaxFunctionalModel=0 +mm_openPagePolicy=1 +mm_backendLatency=2 +mm_schedulerWindowSize=8 +mm_transactionQueueDepth=8 +mm_dramTimings_tAL=0 +mm_dramTimings_tCAS=14 +mm_dramTimings_tCMD=1 +mm_dramTimings_tCWD=10 +mm_dramTimings_tCCD=4 +mm_dramTimings_tFAW=25 +mm_dramTimings_tRAS=33 +mm_dramTimings_tREFI=7800 +mm_dramTimings_tRC=47 +mm_dramTimings_tRCD=14 +mm_dramTimings_tRFC=160 +mm_dramTimings_tRRD=8 +mm_dramTimings_tRP=14 +mm_dramTimings_tRTP=8 +mm_dramTimings_tRTRS=2 +mm_dramTimings_tWR=15 +mm_dramTimings_tWTR=8 +mm_rowAddr_offset=18 +mm_rowAddr_mask=65535 +mm_rankAddr_offset=16 +mm_rankAddr_mask=3 +mm_bankAddr_offset=13 +mm_bankAddr_mask=7 +mm_llc_wayBits=3 +mm_llc_setBits=12 +mm_llc_blockBits=7 +mm_llc_activeMSHRs=8 +macaddr0=00:12:6D:00:00:02 +slotid=0 +niclog0=niclog +trace-start0=0 +trace-end0=-1 +linklatency0=6405 +netbw0=200 +profile-interval=-1 +zero-out-dram +shmemportname0=default +permissive-off br-base-bin +blkdev0=br-base.img
random min: 0x0, random max: 0xffffffffffffffff
Collect trace from 0 to 18446744073709551615 cycles
Zeroing out FPGA DRAM. This will take a few seconds...
```

And there is no further response. 

The simulation status from manager seem all right, is there anything I could possible go wrong? Thank you.
```

FireSim Simulation Status @ 2019-11-13 03:05:21.976339
--------------------------------------------------------------------------------
This workload's output is located in:
/home/centos/firesim/deploy/results-workload/2019-11-13--03-03-33-linux-uniform/
This run's log is located in:
/home/centos/firesim/deploy/logs/2019-11-13--03-03-33-runworkload-U8RUI5WHF61E7WPB.log
This status will update every 10s.
--------------------------------------------------------------------------------
Instances
--------------------------------------------------------------------------------
Instance IP:   192.168.4.78 | Terminated: False
--------------------------------------------------------------------------------
Simulated Switches
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
Simulated Nodes/Jobs
--------------------------------------------------------------------------------
Instance IP:   192.168.4.78 | Job: linux-uniform0 | Sim running: True 
--------------------------------------------------------------------------------
Summary
--------------------------------------------------------------------------------
1/1 instances are still running.
1/1 simulations are still running.
--------------------------------------------------------------------------------
```

davidbiancolin, 2019-11-13:
Alright so, the CPU usage should be low for buildafi -- once the RTL is generated the build is batched out to another AWS instance.

Can you attach your infrasetup log? The manager should dump a bunch of log files into deploy/logs.

Yangff, 2019-11-13:
> Alright so, the CPU usage should be low for buildafi -- once the RTL is generated the build is batched out to another AWS instance.
> 
> Can you attach your infrasetup log? The manager should dump a bunch of log files into deploy/logs.

buildafi didn't start any instance and also not run anything locally.. for about 20 minutes.. 

These logs for `infrasetup` from tag 1.7.0, it's a bit large so I attached it. 

[2019-11-13--08-25-17-infrasetup-JH5VS5U1HYLZKMMG.log](https://github.com/firesim/firesim/files/3842898/2019-11-13--08-25-17-infrasetup-JH5VS5U1HYLZKMMG.log)

davidbiancolin, 2019-11-13:
Can you provide your buildafi log too?

Yangff, 2019-11-13:
> Can you provide your buildafi log too?

Nothing in that log.. but if you want:

```
centos@ip-192-168-5-102.ec2.internal:~/firesim/deploy/logs$ cat 2019-11-13--09-12-40-buildafi-F4QBILFLVXUR5MVE.log
2019-11-13 09:12:40,085 [main        ] [INFO ]  FireSim Manager. Docs: http://docs.fires.im
Running: buildafi

2019-11-13 09:12:40,092 [aws_resource] [DEBUG]  i-0eddd97e0dfd6f077
2019-11-13 09:12:40,270 [aws_resource] [DEBUG]  {}
2019-11-13 09:12:40,332 [flush       ] [DEBUG]  [localhost] local: pwd
2019-11-13 09:12:40,336 [replace_rtl ] [INFO ]  Running replace-rtl to generate verilog for FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz
2019-11-13 09:12:40,336 [flush       ] [DEBUG]  [localhost] run: make  DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig PLATFORM_CONFIG=BaseF1Config_F90MHz replace-rtl
2019-11-13 10:07:35,040 [flush       ] [DEBUG]  [localhost] Login password for 'centos': 
2019-11-13 10:07:35,040 [flush       ] [DEBUG]  Sorry, you can't enter an empty password. Please try again.
2019-11-13 10:07:35,497 [flush       ] [DEBUG]  [localhost] Login password for 'centos': 
2019-11-13 10:07:35,498 [flush       ] [DEBUG]  Sorry, you can't enter an empty password. Please try again.
2019-11-13 10:07:35,807 [flush       ] [DEBUG]  [localhost] Login password for 'centos': 
2019-11-13 10:07:35,807 [flush       ] [DEBUG]  Sorry, you can't enter an empty password. Please try again.
2019-11-13 10:07:36,613 [terminate_in] [INFO ]  You pressed ctrl-c, so builds have been killed.
2019-11-13 10:07:36,613 [flush       ] [DEBUG]  Do you also want to terminate your build instances? Type 'yes' to do so.
2019-11-13 10:07:42,575 [terminate_in] [INFO ]  Termination skipped. There may still be build instances running.
2019-11-13 10:07:42,575 [<module>    ] [ERROR]  Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 311, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 246, in main
    globals()[args.task](buildconfig)
  File "/home/centos/firesim/deploy/firesim", line 113, in buildafi
    execute(replace_rtl, globalbuildconf, buildconf, hosts=['localhost'])
  File "/usr/lib64/python2.7/site-packages/fabric/tasks.py", line 386, in execute
    multiprocessing
  File "/usr/lib64/python2.7/site-packages/fabric/tasks.py", line 276, in _execute
    return task.run(*args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/fabric/tasks.py", line 173, in run
    return self.wrapped(*args, **kwargs)
  File "/home/centos/firesim/deploy/buildtools/buildafi.py", line 34, in replace_rtl
    run(buildconfig.make_recipe("replace-rtl"))
  File "/usr/lib64/python2.7/site-packages/fabric/network.py", line 687, in host_prompting_wrapper
    return func(*args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/fabric/operations.py", line 1090, in run
    shell_escape=shell_escape, capture_buffer_size=capture_buffer_size,
  File "/usr/lib64/python2.7/site-packages/fabric/operations.py", line 930, in _run_command
    channel=default_channel(), command=wrapped_command, pty=pty,
  File "/usr/lib64/python2.7/site-packages/fabric/state.py", line 435, in default_channel
    chan = _open_session()
  File "/usr/lib64/python2.7/site-packages/fabric/state.py", line 416, in _open_session
    transport = connections[env.host_string].get_transport()
  File "/usr/lib64/python2.7/site-packages/fabric/network.py", line 159, in __getitem__
    self.connect(key)
  File "/usr/lib64/python2.7/site-packages/fabric/network.py", line 151, in connect
    user, host, port, cache=self, seek_gateway=seek_gateway)
  File "/usr/lib64/python2.7/site-packages/fabric/network.py", line 569, in connect
    password = prompt_for_password(text)
  File "/usr/lib64/python2.7/site-packages/fabric/network.py", line 657, in prompt_for_password
    new_password = _password_prompt(password_prompt, stream)
  File "/usr/lib64/python2.7/site-packages/fabric/network.py", line 624, in _password_prompt
    return getpass.getpass(prompt.encode('ascii', 'ignore'), stream)
  File "/usr/lib64/python2.7/getpass.py", line 71, in unix_getpass
    passwd = _raw_input(prompt, stream, input=input)
  File "/usr/lib64/python2.7/getpass.py", line 133, in _raw_input
    line = input.readline()
  File "/home/centos/firesim/deploy/firesim", line 108, in terminate_instances_handler
    sys.exit(0)
SystemExit: 0
2019-11-13 10:07:42,576 [<module>    ] [INFO ]  The full log of this run is:
/home/centos/firesim/deploy/logs/2019-11-13--09-12-40-buildafi-F4QBILFLVXUR5MVE.log
centos@ip-192-168-5-102.ec2.internal:~/firesim/deploy/logs$ 
```

Yangff, 2019-11-13:
And.. If this will help, I can add your public key to my AWS server.. I feel it may has something to do with the FPGA side instead of manager server...

davidbiancolin, 2019-11-14:
There's seems to be something weird going on in buildafi. Are you forwarding your ssh-agent when you log on to your manager?

Yangff, 2019-11-14:
> There's seems to be something weird going on in buildafi. Are you forwarding your ssh-agent when you log on to your manager?

I'm using Xshell and I don't think it forwards ssh-agent..

davidbiancolin, 2019-11-14:
```
2019-11-13 09:12:40,336 [flush       ] [DEBUG]  [localhost] run: make  DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig PLATFORM_CONFIG=BaseF1Config_F90MHz replace-rtl
2019-11-13 10:07:35,040 [flush       ] [DEBUG]  [localhost] Login password for 'centos': 
2019-11-13 10:07:35,040 [flush       ] [DEBUG]  Sorry, you can't enter an empty password. Please try again.
```

This generally happens because the user is agent forwarding. I'm assuming if you run 
`make  DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig PLATFORM_CONFIG=BaseF1Config_F90MHz replace-rtl`
in sim/ it works fine?

Yangff, 2019-11-14:
> ```
> 2019-11-13 09:12:40,336 [flush       ] [DEBUG]  [localhost] run: make  DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig [...]

This happens after I terminate the execution through Ctrl+C.. and i think it's bacause I set password for local `centos` account.. because after I delete the password for `centos` it won't have this problem..

I'll try the command and post the result in minute

Yangff, 2019-11-14:
> make DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig PLATFORM_CONFIG=BaseF1Config_F90MHz replace-rtl

```
centos@ip-192-168-5-102.ec2.internal:~/firesim/sim$ make DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig PLATFORM_CONFIG=BaseF1Config_F90MHz replace-rtl
/home/centos/firesim/sim/src/main/cc/firesim/firesim_top.cc /home/centos/firesim/sim/src/main/cc/firesim/systematic_scheduler.cc /home/centos/firesim/sim/src/main/cc/firesim/firesim_f1.cc /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/uart.cc /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/blockdev.cc /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/serial.cc /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/simplenic.cc /home/centos/firesim/sim/firesim-lib/src/main/cc/bridges/tracerv.cc /home/centos/firesim/sim/firesim-lib/src/main/cc/fesvr/firesim_fesvr.cc /home/centos/firesim/target-design/chipyard/riscv-tools-install/lib/libfesvr.a
mkdir -p /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz
cp -rf /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim -T /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz
touch /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/stamp
/home/centos/firesim/sim/../scripts/repo_state_summary.sh > /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/repo_state
cp -f /home/centos/firesim/sim/generated-src/f1/FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/FPGATop.v /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/cl_firesim_generated.sv
sed -i "s/\$random/64'b0/g" /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/cl_firesim_generated.sv
sed -i 's/fatal/fatal(0, "")/g' /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/cl_firesim_generated.sv
cp -f /home/centos/firesim/sim/generated-src/f1/FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/firesim_ila_insert_* /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/ila_files/
sed -i "s/\$random/64'b0/g" /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/ila_files/*
sed -i 's/fatal/fatal(0, "")/g' /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/ila_files/*
cp -f /home/centos/firesim/sim/generated-src/f1/FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/cl_firesim_generated_defines.vh /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/cl_firesim_generated_defines.vh
cp -f /home/centos/firesim/sim/generated-src/f1/FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/cl_firesim_generated_env.tcl /home/centos/firesim/sim/../platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_FireSimNoNIC-L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimRocketChipQuadCoreConfig-BaseF1Config_F90MHz/design/cl_firesim_generated_env.tcl
centos@ip-192-168-5-102.ec2.internal:~/firesim/sim$ 

```

It seems to work

davidbiancolin, 2019-11-14:
How many different builds are you trying to run in your config_build.ini?

Yangff, 2019-11-14:
> How many different builds are you trying to run in your config_build.ini?

only `firesim-quadcore-no-nic-l2-llc4mb-ddr3`

davidbiancolin, 2019-11-14:
Did you copy your firesim.pem to your manager instance, and is it exactly the same key you associated with the manager instance when you launched it?

Yangff, 2019-11-14:
> Did you copy your firesim.pem to your manager instance, and is it exactly the same key you associated with the manager instance when you launched it?

sure.. since I can ssh to my FPGA instance created by firesim... I believe it's the same..

Yangff, 2019-11-14:
emm. wait.. you mean the manager instance?

Yangff, 2019-11-14:
ok.. I added the public key of `firesim` and it's running now..

davidbiancolin, 2019-11-14:
Yeah so, if you use a different key, say `manager.pem` to ssh onto the manager than `firesim.pem`, which the manager will use to interact with other instances, you'll have problems. 

When the manager runs some tasks on the localhost, paramiko (the library it's using under-the-hood) essentially tries to ssh on to localhost and does so using your firesim.pem key to authenticate.

davidbiancolin, 2019-11-14:
It was probably just waiting for a password on prompt until you you ^-C -ed it.

Yangff, 2019-11-14:
> It was probably just waiting for a password on prompt until you you ^-C -ed it.

understand.. so now it shows 
```
[192.168.2.130] Executing task 'instance_liveness'
[192.168.2.130] Checking if host instance is up...
```
It that means buildafi is running on another instance normally?

davidbiancolin, 2019-11-14:
Yeah

Yangff, 2019-11-14:
Try `firesim runworkload` again and still same result.. only get `Zeroing out FPGA DRAM. This will take a few seconds...` this  time...

davidbiancolin, 2019-11-14:
Despite the message, it should take a couple minutes.

Yangff, 2019-11-14:
> Despite the message, it should take a couple minutes.

it's been five minutes..

Yangff, 2019-11-14:
and it boot at six minutes.!!

davidbiancolin, 2019-11-14:
No problem. Good luck!

Yangff, 2019-11-14:
By the way.. could you tell me about who is providing device tree for bbl..? Will the simulator manage it or I need to hardcode it if I want to boot something else without built-in device tree?

Yangff, 2019-11-15:
Is there any limitation on the bootloader? like image size (~40mb because ramdisk is included) or bbl version? I'm trying to boot other program.. but it's using bbl built from `riscv/riscv-pk` and.. it's not booting again..

I use the tracev to look at the PC, found that the core0 is stop at 0x1005c (`wfi`) and simply waiting for interruption... which sounds to me the fesrv side isn't loading the program as it suppose to? (or loaded but not send the interruption for some reason.. whatever, the result is the same...)

I search in the code tree but cannot find the source code of `htif` or `elfloader.cc` used by sim... 

I'll later try to make a smaller bbl..

davidbiancolin, 2019-11-20:
We've loaded very large images (100s of MB) without issue. 

Any luck?

Yangff, 2019-11-20:
> We've loaded very large images (100s of MB) without issue.
> 
> Any luck?

seems caused by the bootloader. I let it log for longer cycles and get some output. Trying to figure out why it's not output bbl's logo..

davidbiancolin, 2019-11-20:
Does the same image run fine in target-level simulation? (Directly on a VCS simulator of your rocket-chip design)

Yangff, 2019-11-20:
> Does the same image run fine in target-level simulation? (Directly on a VCS simulator of your rocket-chip design)

Havn't tried.. what I get from the log is the core 0's bbl is running `boot_other_hart`..

Two problems.. 1. why 2. it should have print its logo before go into `boot_other_hart`..

At some point, core 0 starts running at `Core 0, Valid=1, Addr=0x80000000; Ins=0x6f602` and after about 400 instructions, it run into `0x800002f_`  and start looping. 

```
0x800002f0      auipc a5, 0x15
0x800002f4      addi a5, a5, -752
0x800002f8      ld a0, 0(a5)
0x800002fa      fence
0x800002fe      beqz a0, 0x800002f8
```
I've no idea why it's looping here...

Yangff, 2019-12-17:
@davidbiancolin 

What I thought is core 0 is actually not.. But now I have another problem..

What I found right now is that `riscv-pk`'s uart is not working on firesim. This issue happens for both `fireism/riscv-pk` and `riscv-pk/riscv-pk`. 

Sometime it's just fine, although not printing anything. 
Sometime It's looping inside `uart_putchar`.. and never return. 
```
    int32_t r;
    do {
      __asm__ __volatile__ (
        "amoor.w %0, %2, %1\n"
        : "=r" (r), "+A" (uart[UART_REG_TXFIFO])
        : "r" (ch));  // <-- looping here
    } while (r < 0);
```

davidbiancolin, 2019-12-18:
Sorry about that; in general you want to use the pk we ship with the version of FireSim you're using until ready to venture off into your own design. There are a couple little modifications we had to make to the target-software to get things running correctly under the default frequencies we are using in our systems. @sagark might be able to provide some more insight.

sagark, 2019-12-23:
As @davidbiancolin mentioned, whenever modifying submodules inside firesim, you should fork off from the submodule commit that firesim is using, which may not necessarily be master of the submoduled repo.

Since it sounds like this is fixed for now, I am closing the issue. Please let us know if it arises again.

closed 2019-12-23 by sagark