firesim/firesim #442: Fatal error: local() encountered an error (return code 255)

CristinaZN, 2019-12-10:
how to solve this problem, please help!

2019-12-09 21:24:38,819 [aws_build ] [DEBUG]
2019-12-09 21:24:38,820 [flush ] [DEBUG] [localhost] local: aws ec2 create-fpga-image --input-storage-location Bucket=firesim-cristina,Key=dcp/19_12_09-083351.Developer_CL.tar-192.168.0.208-ITGFH8RO7L.tar --logs-storage-location Bucket=firesim-cristina,Key=logs/ --name "firesim-quadcore-no-nic-nvdla-ddr3-llc4mb" --description "firesim-buildtriplet:FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig75MHz,firesim-deploytriplet:FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig75MHz,firesim-commit:31357f963f875fc2ee37045c8655a68a276b5ff7-dirty"
2019-12-09 21:24:39,655 [flush ] [DEBUG] Fatal error: local() encountered an error (return code 255) while executing 'aws ec2 create-fpga-image --input-storage-location Bucket=firesim-cristina,Key=dcp/19_12_09-083351.Developer_CL.tar-192.168.0.208-ITGFH8RO7L.tar --logs-storage-location Bucket=firesim-cristina,Key=logs/ --name "firesim-quadcore-no-nic-nvdla-ddr3-llc4mb" --description "firesim-buildtriplet:FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig75MHz,firesim-deploytriplet:FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig75MHz,firesim-commit:31357f963f875fc2ee37045c8655a68a276b5ff7-dirty" '
2019-12-09 21:24:39,655 [flush ] [DEBUG] Aborting.
2019-12-09 21:24:39,678 [ ] [ERROR] Fatal error.

davidbiancolin, 2019-12-10:
Hey Cristina,

Does the bucket exist? Can you try uploading something to it? Ideally, if you still had the build instance running you should try rerunning the command from there.

CristinaZN, 2019-12-11:
hello, i had stop the build instance..... so i do `firesim buildafi` again. but get another ERROR:
upload: ./19_12_10-131852.Developer_CL.tar to s3://firesim-cristina/dcp/19_12_10-131852.Developer_CL.tar-192.168.0.119-ESSUMLL9Y9.tar
2019-12-11 03:02:21,345 [aws_build   ] [DEBUG]  
2019-12-11 03:02:21,345 [flush       ] [DEBUG]  [localhost] local: aws ec2 create-fpga-image --input-storage-location Bucket=firesim-cristina,Key=dcp/19_12_10-131852.Developer_CL.tar-192.168.0.119-ESSUMLL9Y9.tar --logs-storage-location Bucket=firesim-cristina,Key=logs/ --name "firesim-quadcore-no-nic-nvdla-ddr3-llc4mb" --description "firesim-buildtriplet:FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig75MHz,firesim-deploytriplet:FireSimNoNIC-FireSimRocketChipQuadCoreConfig_WithNVDLALarge-FireSimDDR3FRFCFSLLC4MBConfig75MHz,firesim-commit:31357f963f875fc2ee37045c8655a68a276b5ff7-dirty" 
2019-12-11 03:02:22,308 [aws_build   ] [DEBUG]  -----------------------------------------------------
|                  CreateFpgaImage                  |
+-------------------------+-------------------------+
|    FpgaImageGlobalId    |       FpgaImageId       |
+-------------------------+-------------------------+
|  agfi-087a2a8a100902ba6 |  afi-083c916adec9f6218  |
+-------------------------+-------------------------+
2019-12-11 03:02:22,308 [aws_build   ] [DEBUG]  
2019-12-11 03:02:22,308 [aws_build   ] [DEBUG]  create-fpge-image result: -----------------------------------------------------
|                  CreateFpgaImage                  |
+-------------------------+-------------------------+
|    FpgaImageGlobalId    |       FpgaImageId       |
+-------------------------+-------------------------+
|  agfi-087a2a8a100902ba6 |  afi-083c916adec9f6218  |
+-------------------------+-------------------------+
2019-12-11 03:02:22,336 [<module>    ] [ERROR]  Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim-nvdla/deploy/firesim", line 307, in <module>
    main(args)
  File "/home/centos/firesim-nvdla/deploy/firesim", line 242, in main
    globals()[args.task](buildconfig)
  File "/home/centos/firesim-nvdla/deploy/firesim", line 122, in buildafi
    execute(aws_build, globalbuildconf, hosts=globalbuildconf.get_build_instance_ips())
  File "/usr/lib64/python2.7/site-packages/fabric/tasks.py", line 418, in execute
    error(err, exception=d['results'])
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 358, in error
    return func(message)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 62, in abort
    raise e
SystemExit: 1

sagark, 2019-12-16:
Closing this since it looks like it was resolved here: https://github.com/CSL-KU/firesim-nvdla/issues/11

Let us know if you're still having issues

closed 2019-12-16 by sagark