firesim/firesim #455: Spec17-intspeed Benchmarks failing for fireboom

jfustos, 2020-01-09:
Hello,

I am working with BOOM and Firesim, trying to run Spec17-intspeed benchmarks.

I am working from commits:
    Firesim: commit 1a3cdb597617fdccb2fcfb8993e468183931853c
    BOOM: commit 2a0ea2e7acfd4605eed513e15062848e4e5be309

My problem is that I keep getting errors such as:
    ./run.sh: line 4:   218 Segmentation fault      ./omnetpp_s_base.riscv-64 -c General -r 0

These seem to happen randomly(not in the same spot) running the same benchmark on the same build and happen in different benchmarks when changing BOOM sizes. If they would happen in the same spot, then I could try and debug them. Example, I am running using build configuaration:
    DESIGN=FireSimNoNIC
    TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFSLLC4MB_FireSimBoomConfigLarge
    PLATFORM_CONFIG=BaseF1Config_F75MHz

Note: FireSimBoomConfigLarge === FireSimBoomConfig I am just building with different sizes and want to keep the outputs strait.

Note: I have tried both building spec with and without compressed instructions

Running 620.omnetpp_s round 1:
    Starting speed 620.omnetpp_s run with 1 threads^M^M
    [    2.038197] omnetpp_s_base.[218]: unhandled signal 11 code 0x1 at 0x0000000000000000 in omnetpp_s_base.riscv-64[10000+378000]^M^M
    [    2.038221] CPU: 0 PID: 218 Comm: omnetpp_s_base. Tainted: G           O      5.3.0-00002-g6a95b016aba5 #1^M^M
    [    2.038237] sepc: 00000000000988d4 ra : 00000000000988d4 sp : 0000003fffac06d0^M^M
    [    2.038388]  gp : 0000000000395fb8 tp : 00000000003a7710 t0 : 0000000000000000^M^M
    [    2.038613]  t1 : 0000000000000000 t2 : 00000000000003ff s0 : 00000000072e1770^M^M
    [    2.038839]  s1 : 0000000002c2d6f0 a0 : 0000000000000000 a1 : 000000000029d301^M^M
    [    2.039064]  a2 : 0000000000000000 a3 : 0000000000000009 a4 : 00000000002ad29b^M^M
    [    2.039290]  a5 : 0000000000028964 a6 : 0000000000000000 a7 : 0000000002c2d670^M^M
    [    2.039515]  s2 : 00000000002ad29a s3 : 0000000006184160 s4 : 0000003fffac0778^M^M
    [    2.039741]  s5 : 00000000002f1c98 s6 : 00000000002aa000 s7 : 0000000000288000^M^M
    [    2.039966]  s8 : 0000000000288000 s9 : 00000000002aa000 s10: 00000000002aa000^M^M
    [    2.040191]  s11: 0000000000000000 t3 : 00000000024f4190 t4 : 00000000024f18e0^M^M
    [    2.040426]  t5 : 00000000024f2d50 t6 : 0000000000000000^M^M
    [    2.040583] sstatus: 8000000200006020 sbadaddr: 0000000000000000 scause: 000000000000000d^M^M
    [    2.053574] reboot: Power down^M^M
    Power off^M
    ^M
    *** PASSED *** after **7327416092** cycles^M

Running 620.omnetpp_s round 2:
    Starting speed 620.omnetpp_s run with 1 threads^M^M
    [   28.887783] omnetpp_s_base.[218]: unhandled signal 4 code 0x1 at 0x00000000002f1c98 in omnetpp_s_base.riscv-64[10000+378000]^M^M
    [   28.887809] CPU: 0 PID: 218 Comm: omnetpp_s_base. Tainted: G           O      5.3.0-00002-g6a95b016aba5 #1^M^M
    [   28.887825] sepc: 00000000002f1c98 ra : 00000000002f1c98 sp : 0000003fffb667c0^M^M
    [   28.887972]  gp : 0000000000395fb8 tp : 00000000003a7710 t0 : 0000000000000001^M^M
    [   28.888198]  t1 : 0000000000000000 t2 : 00000000000003ff s0 : 000000000078a7a0^M^M
    [   28.888434]  s1 : 0000003fffb66838 a0 : 0000000000000000 a1 : 00000000031486e0^M^M
    [   28.888648]  a2 : 0000003fffb667a8 a3 : 0000000000000000 a4 : 00000007c6c30280^M^M
    [   28.888874]  a5 : 0000000000000000 a6 : 00000000003a7ff0 a7 : 0000000000000007^M^M
    [   28.889099]  s2 : 00000000003edb70 s3 : 0000000000397000 s4 : 0000003fffb669f0^M^M
    [   28.889325]  s5 : 00000000002f1c98 s6 : 00000000002aa000 s7 : 0000000000288000^M^M
    [   28.889550]  s8 : 0000000000288000 s9 : 00000000002aa000 s10: 00000000002aa000^M^M
    [   28.889776]  s11: 0000000000000000 t3 : 0000000006da3208 t4 : 00000007c9f9c380^M^M
    [   28.890001]  t5 : 0000000000ce3095 t6 : 0000000000000000^M^M
    [   28.890167] sstatus: 8000000200006020 sbadaddr: 0000000000000000 scause: 0000000000000002^M^M
    [   28.910363] reboot: Power down^M^M
    Power off^M
    ^M
    *** PASSED *** after **93265677347** cycles^M

The runs fail on different cycles and at different points in the program?
I have also built with asserts enabled, and faults do not cause asserts.

This seems to be happening on both small and large Booms, even though it seems to be happening more on the Large Booms(Mega Boom will not even boot linux, but does have the following assert which may be unrelated to the problem, and I have not tested on Rocket Core)

    [id: 945, module: LSU, path: FireSimNoNIC.targets.boom_tile.lsu]^M
    Assertion failed^M
        at lsu.scala:1259 assert(ldq(io.dmem.nack(w).bits.uop.ldq_idx).bits.executed)^M
     at cycle: 4009614^M

Any suggestions?
Is there a stable commit and configuration that I should be using for running Spec17?

jerryz123, 2020-01-09:
Please use the latest version of BOOM, several memory-related bugs were fixed between the version you have and the latest version.

jfustos, 2020-01-20:
Thank you for the help @jerryz123 . I was able to switch to dev on firesim and pull up to the latest commit for Boom and the memory errors are gone. You are a life saver.

closed 2020-01-20 by jfustos