firesim/firesim #484: TracerV issue with multicore BOOM

farzadfch, 2020-01-30:
This might be already a known issue but towards the end of the GG elaboration of multicore BOOM designs, it complains about TracerV. Removing [this line](https://github.com/ucb-bar/chipyard/blob/10422488a24872301b02a80d3a4dbbe9cca91e36/generators/firechip/src/main/scala/BridgeBinders.scala#L81) temporarily fixes the issue.

davidbiancolin, 2020-01-30:
Can you provide your stack trace?

davidbiancolin, 2020-01-30:
My guess is that the combined bit width of all the trace ports exceeds 512 bits

farzadfch, 2020-01-30:
[2020-01-28--22-16-40-buildafi-F2BBO9GVGBEA21Z2.log](https://github.com/firesim/firesim/files/4136408/2020-01-28--22-16-40-buildafi-F2BBO9GVGBEA21Z2.log)
This is the log for a custom BOOM but the issue manifests with out-of-box configurations too.

davidbiancolin, 2020-01-30:
This is 1.8.0 right?

farzadfch, 2020-01-30:
Yes.

farzadfch, 2020-01-31:
Sure,
DESIGN=FireSimNoNIC TARGET_CONFIG=L2SingleBank512K_DDR3FRFCFS_FireSimBoomQuadCoreConfig PLATFORM_CONFIG=BaseF1Config

davidbiancolin, 2020-02-01:
I've locally reproduced this and i've found that if the total commit bandwidth of the machine is too high TracerV conks out because it can't pack 4 (cores) x 3 (instructions) x 64 bits into a single 512 token. 

In our multiclock branch we've accidentally fixed this by using multiple tracerV bridges (one per core) which would require a 7-wide machine to break. What i propose we do is just truncate the trace to available width until the multi-bridge implementation is introduced.

closed 2020-06-01 by davidbiancolin

mbelda, 2021-12-10:
Hello, 
I am having issues building an AGFI with the newer version of the repo for a BOOM 4-core system.
It throws an error with little information. 
I've tried to follow this issue but anyway it gives the error.

My config is the following:
[firesim-boom-quadcore-no-nic-l2-llc4mb-ddr3-MCRams-50MHz]
DESIGN=FireSim
TARGET_CONFIG=DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.QuadLargeBoomConfig
PLATFORM_CONFIG=MCRams_F50MHz_BaseF1Config
instancetype=z1d.2xlarge
deploytriplet=None

I have also tried adding WithAutoILA. And I have successfully generated other architectures with 4-Rocket multicore, so i think is not a problem of the environment.

And my log file for the run can be find here [https://drive.google.com/file/d/1MI_7e-_9ypgVsa6cAVCK0MB2pTEjDy-x/view?usp=sharing](url)

PR #529 (2020-06-01): FireSim 1.10 Dev to Master Tracking PR 
**Developers: Please add descriptions of your changes to this PR as your PRs are merged into Dev**

### Added
* Support for simulating targets with multiple fixed-frequency clock (MC) domains (PR #441)
  * All clocks must be generated using the RationalClockBridge
  * See docs (PR #527)
* Generalized trigger system (part of MC, PR #441) (resolves #497)
  * See docs (PR #526)
* Intelligent Bridge DRAM allocation (PR #433)
  * Mix in `UsesHostDRAM` into a bridge that needs FPGA-DRAM
* CircleCI  integration (PR #534, PR #574)
* ScalaDoc for dev branch autom [... cut]