firesim/firesim #505: How slow firesim will be when the TracerV is enable？

lightningwm, 2020-02-15:
I want to track the instruction stream by TracerV, while the document does't describe enough detail about it. If I use TracerV, how slow the speed of simulator will become？

davidbiancolin, 2020-02-15:
Hey lightningwm,

That depends on a few factors, namely: 
- how many cores you have in your target
- what print format you choose 
- host-FPGA frequency 
- how many host-stalls other bridges in your system are introducing

Fully tracing 4 rocket cores using the human readable format runs about 12x slower. 

Also, in the future i'd encourage you to post questions of this nature to our [google group](https://groups.google.com/forum/#!forum/firesim).

Kenun99, 2020-02-16:
Hi, David

Thanks for your reply.

davidbiancolin, 2020-04-02:
Closing, as this has been discussed on the google group. Feel free to re-open if you still have questions.

closed 2020-04-02 by davidbiancolin