firesim/firesim #509: Farm out chisel build process to build nodes

sagark, 2020-02-18:
Running these locally during a buildafi becomes problematic when running lots of configs, esp for BOOM.

I got part of the way there with:

rsync -zarv --include="*/" --include="*.sbt" --include="*.scala" --include="*Makefile" --include="Makefrag" --include="*.mk" --include="*.jar" --exclude="*" firesim-hyper REMOTE_IP:

And, installing dtc, java on the remote nodes. But eventually still errors during elab. Will try again later.

davidbiancolin, 2020-04-02:
i think the first thing we can do make this less painful is to parallelize it locally

timsnyder-siv, 2020-10-15:
> i think the first thing we can do make this less painful is to parallelize it locally

I've been doing this but given the parameters that I have set for GoldenGate, I tend to run out of RAM on my manager node (perhaps I need to tune the JVM RAM back).  

Rather than having to copy all of the scala collateral over, compile the scala into an assembly FAT JAR once and copy that over to all the build hosts.  I'm looking at FAT JAR support in #646