firesim/firesim #522: Support BlackBox RTL designs containing include directives

tymcauley, 2020-03-16:
I'm using FireSim as a library from Chipyard (release 1.2.0), and I'm attempting to get a large multi-file Verilog design to integrate with FireSim. This Verilog design uses a few `.vh` files, and many of the source `.v` files use `include` directives to include those `.vh` files.

Commit 16abe872870871f9739f27c0651629600bf2f0ab added this change to `sim/target-agnostic.mk`:

```diff
diff --git a/sim/target-agnostic.mk b/sim/target-agnostic.mk
index e2a41519..86005f71 100644
--- a/sim/target-agnostic.mk
+++ b/sim/target-agnostic.mk
@@ -55,6 +55,8 @@ $(VERILOG) $(HEADER): $(FIRRTL_FILE) $(ANNO_FILE)
                -ggcp $(PLATFORM_CONFIG_PACKAGE) \
                -ggcs $(PLATFORM_CONFIG) \
                -E verilog"
+       grep -sh ^ $(GENERATED_DIR)/firrtl_black_box_resource_files.f | \
+       xargs cat >> $(VERILOG) # Append blackboxes to FPGA wrapper, if any

 ####################################
 # Runtime-Configuration Generation #
```

This appends every single file listed in the `firrtl_black_box_resource_files.f` file to the top-level verilog file generated by the FIRRTL compiler. This presents a problem: any `include` directives won't find their desired files, since all the blackbox source is just inlined into a single file.

We could solve this issue by instead copying all files in `firrtl_black_box_resource_files.f` to the FPGA build area, using something like this:

```diff
diff --git a/sim/target-agnostic.mk b/sim/target-agnostic.mk
index 1b782396..6ed9e363 100644
--- a/sim/target-agnostic.mk
+++ b/sim/target-agnostic.mk
@@ -55,8 +55,6 @@ $(VERILOG) $(HEADER): $(FIRRTL_FILE) $(ANNO_FILE)
                -ggcp $(PLATFORM_CONFIG_PACKAGE) \
                -ggcs $(PLATFORM_CONFIG) \
                -E verilog"
-       grep -sh ^ $(GENERATED_DIR)/firrtl_black_box_resource_files.f | \
-       xargs cat >> $(VERILOG) # Append blackboxes to FPGA wrapper, if any

 ####################################
 # Runtime-Configuration Generation #
@@ -163,6 +161,7 @@ ila_work_dir   := $(fpga_work_dir)/design/ila_files/
 fpga_vh        := $(fpga_work_dir)/design/cl_firesim_generated_defines.vh
 fpga_tcl_env   := $(fpga_work_dir)/design/cl_firesim_generated_env.tcl
 repo_state     := $(fpga_work_dir)/design/repo_state
+blackbox_dir   := $(fpga_work_dir)/design/blackbox

 $(fpga_work_dir)/stamp: $(shell find $(board_dir)/cl_firesim -name '*')
        mkdir -p $(@D)
@@ -187,9 +186,14 @@ $(ila_work_dir): $(verilog) $(fpga_work_dir)/stamp
        sed -i "s/\$$random/64'b0/g" $(fpga_work_dir)/design/ila_files/*
        sed -i "s/\(^ *\)fatal;\( *$$\)/\1fatal(0, \"\");\2/g" $(fpga_work_dir)/design/ila_files/*

+.PHONY: copy-blackbox-rtl
+copy-blackbox-rtl: $(VERILOG) $(fpga_work_dir)/stamp
+       mkdir -p $(blackbox_dir)
+       xargs -a $(GENERATED_DIR)/firrtl_black_box_resource_files.f cp -t $(blackbox_dir)
+
 # Goes as far as setting up the build directory without running the cad job
 # Used by the manager before passing a build to a remote machine
-replace-rtl: $(fpga_v) $(ila_work_dir) $(fpga_vh) $(fpga_tcl_env)
+replace-rtl: $(fpga_v) $(ila_work_dir) $(fpga_vh) $(fpga_tcl_env) copy-blackbox-rtl
```

Then you must also adjust `platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/encrypt.tcl` and `platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl` to ensure that these files are read appropriately:

```diff
Submodule platforms/f1/aws-fpga contains modified content
diff --git a/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/encrypt.tcl b/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/encrypt.tcl
index e5630bc..0fc6bee 100644
--- a/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/encrypt.tcl
+++ b/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/encrypt.tcl
@@ -53,6 +53,12 @@ file copy -force $UNUSED_TEMPLATES_DIR/unused_pcim_template.inc       $TARGET_DI
 file copy -force $UNUSED_TEMPLATES_DIR/unused_sh_bar1_template.inc    $TARGET_DIR
 file copy -force $UNUSED_TEMPLATES_DIR/unused_flr_template.inc        $TARGET_DIR

+# Include any blackbox RTL files
+set blackbox_files [glob -nocomplain $CL_DIR/design/blackbox/*]
+foreach f $blackbox_files {
+  file copy -force $f $TARGET_DIR
+}
+
 #---- End of section replaced by Developr ---

diff --git a/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl b/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl
index 9490754..b848ba9 100644
--- a/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl
+++ b/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl
@@ -36,7 +36,7 @@ puts "AWS FPGA: ([clock format [clock seconds] -format %T]) Reading developer's
 # Reading the .sv and .v files, as proper designs would not require
 # reading .v, .vh, nor .inc files

-read_verilog -sv [glob $ENC_SRC_DIR/*.?v]
+read_verilog -sv [glob $ENC_SRC_DIR/*.*v*]

 #---- End of section replaced by User ----
```

However, that's not actually the root cause of the problem, because if you look at `firrtl_black_box_resource_files.f`, it doesn't even contain the verilog header files (even though they are copied into the `$(build_dir)` directory). That results from this code from the `firrtl` project in [`src/main/scala/firrtl/transforms/BlackBoxSourceHelper.scala`](https://github.com/freechipsproject/firrtl/blob/44f0112c7a9d9e9fa7f87fa6e5f68916e76a3b19/src/main/scala/firrtl/transforms/BlackBoxSourceHelper.scala#L121-L124)

```scala
// Issue #917 - We don't want to list Verilog header files ("*.vh") in our file list - they will automatically be included by reference.
def isHeader(name: String) = name.endsWith(".h") || name.endsWith(".vh") || name.endsWith(".svh")
val verilogSourcesOnly = (resourceFiles ++ inlineFiles).filterNot{ f => isHeader(f.getName()) }
```

Here's the associated issue mentioned in that code comment: https://github.com/freechipsproject/firrtl/issues/917

So, unless I'm missing something, it doesn't seem like we have a good way of tracking Verilog header files added with `addResource()` annotations, and including them in the FPGA design.

Any ideas on an alternative solution? We could patch `firrtl` so that all of the files matched by that `isHeader()` function are dumped to a separate file we could use.

abejgonzalez, 2020-03-16:
So I had this exact problem when trying to integrate Ariane (https://github.com/pulp-platform/ariane) into Chipyard (https://github.com/ucb-bar/ariane-wrapper/tree/master). The temporary solution that I used was to pre-process the Verilog files with Verilator using this Makefile (https://github.com/ucb-bar/ariane-wrapper/blob/master/src/main/resources/vsrc/Makefile). This in-lines all the includes. So this is one way to deal with this issue (and combine everything into one file).

I think another way is to have a `addHeaderResource(baseDir, pathToFile)` that creates a directory in the `generated-src` that holds all headers in the same directory structure as the included Verilog projects. For example:

Verilog Projects Being Included:
```
projectA/
  src/
  include/
    header1.vh
    header2.vh

projectB/
   inc/
     moduleX-incs/
        header3.vh
```

Using the `addResourceHeader` func:
```
addResourceHeader("include", "PATH/include/header1.vh")
addResourceHeader("include", "PATH/include/header2.vh")
addResourceHeader("inc", "PATH/inc/moduleX-incs/header3.vh")
```

Then this is generated in the `generated-src` area (so it can be included with a `-I$PWD/generated-src/verilog-includes` flag in simulation):
```
generated-src/
  verilog-includes/
     include/
       header1.vh
       header2.vh
     inc/
       moduleX-incs/
          header3.vh
```

In synthesis, maybe you could just copy this "verilog-includes" and point to it in the `tcl` with an equivalent `-I` flag/command. 

However, this is just a step towards copying all the Verilog sources into `generated-src`.

Of course, this is just brainstorming so I think this doesn't cover all cases. Maybe this sparks some better ideas?

tymcauley, 2020-03-19:
Thanks for the ideas, and the quick response! So, it looks like the ariane codebase extensively uses SystemVerilog packages, and `import` commands for this (which, for the record, is way better than `include` directives, so way to go 😄). In those situations, the strategy of "`cat` all the files together and run them through your verilog compiler" works fine.

However, files like `$(ariane_dir)/src/fpu/src/fpnew_fma.sv` have lines that look like this:

```verilog
`include "common_cells/registers.svh"
```

It looks like you're adding that file to the search path with this code in your Makefile:

```make
lookup_dirs = $(shell find -L $(ariane_dir) -name target -prune -o -type d -print 2> /dev/null | grep '.*/\($(1)\)$$')
INC_DIR_NAMES ?= include inc
INC_DIRS ?= $(foreach dir_name,$(INC_DIR_NAMES),$(call lookup_dirs,$(dir_name)))
...
PREPROC_VERILATOR_OPTS = \
	...
	$(foreach dir_path,$(INC_DIRS),+incdir+$(dir_path)) \
	...
```

Does that sound like I'm interpreting your code properly?

Have you been able to simulate Ariane in FireSim? It seems like you're creating a Verilator executable in this Makefile, but it seems like you'd have to do this all over again when passing your Verilog collateral to the AWS FPGA subsystem.

Another idea: Just rename every `.vh`/`.svh` source file to `.v`/`.sv`. I'll try that out, see if it gets around some of these issues.

tymcauley, 2020-03-28:
Related issue: I just noticed this issue in `synth_cl_firesim.tcl`:

```tcl
#---- User would replace this section -----

# Reading the .sv and .v files, as proper designs would not require
# reading .v, .vh, nor .inc files

read_verilog -sv [glob $ENC_SRC_DIR/*.?v]

#---- End of section replaced by User ----
```

According to the "globbing patterns" section on [this page](https://www.tcl.tk/man/tcl/TclCmd/glob.htm), the `?` in a `glob` matches any single character. That means this pattern won't match any `.v` files. This change has the intended effect (based on the comment):

```diff
diff --git a/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl b/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl
index 9490754..b848ba9 100644
--- a/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl
+++ b/platforms/f1/aws-fpga/hdk/cl/developer_designs/cl_firesim/build/scripts/synth_cl_firesim.tcl
@@ -36,7 +36,7 @@ puts "AWS FPGA: ([clock format [clock seconds] -format %T]) Reading developer's
 # Reading the .sv and .v files, as proper designs would not require
 # reading .v, .vh, nor .inc files

-read_verilog -sv [glob $ENC_SRC_DIR/*.?v]
+read_verilog -sv [glob $ENC_SRC_DIR/*.{sv,v}]

 #---- End of section replaced by User ----
```

abejgonzalez, 2020-05-13:
Hey Tynan,

Sorry, I didn't see the Mar. 19th/27th post. Even though I think most of this information has been given in other areas, I'll go ahead and answer here for other people who might be interested.

> Does that sound like I'm interpreting your code properly?

Yes. That is correct, at the time of my response, I was using the Verilator pre-processor to replace the includes and I was used the `$(INC_DIRS)` variable to get all search paths.

> Have you been able to simulate Ariane in FireSim?

Yup. Ariane is able to boot `br-base` successfully. I was not building a Verilator executable. The `-E` flag instead just runs the pre-processor. In the most recent work with NVDLA, I ended up removing the need for the Verilator pre-processor and I instead use a script to replace the includes. See https://github.com/ucb-bar/ariane-wrapper/pull/5 and the NVDLA integration PR https://github.com/ucb-bar/chipyard/pull/505. In both cases, you run the "pre-process step" (verilator pre-processor or my script) to get a single combined Verilog file without imports. Once that is created, you shouldn't have to re-run that step unless you need to re-elaborate the entire system.

I don't think your tcl change is necessary. IIRC FireSim combines everything into a single `.sv` file that is given to synthesis. This file includes the blackboxes (Ariane, etc) as well as the other Chisel-generated Verilog. So just matching on the `sv` file is fine.

tymcauley, 2020-05-18:
Sounds great, that all makes sense, thanks for the details @abejgonzalez! Not sure if the nitpick is worth a whole PR, but should we make this change to make this comment less misleading and the code a little clearer?

```
 # Reading the .sv and .v files, as proper designs would not require
 # reading .v, .vh, nor .inc files

read_verilog -sv [glob $ENC_SRC_DIR/*.?v]
```
->
```
 # Reading the .sv files, as proper designs would not require
 # reading .v, .vh, nor .inc files

read_verilog -sv [glob $ENC_SRC_DIR/*.sv]
```

Or is there a use-case for reading files with other extensions that match that `.?v` glob?

abejgonzalez, 2020-05-18:
That change makes sense to me. I don't know of any other files that are `.*v`. Go ahead and PR the change and maybe other FireSim devs can chime in if this was intended or not.

HaFred, 2020-06-19:
@tymcauley Hi, may I know how do you access Firesim? Have you purchased the AWS in this case? Thank you very much.

tymcauley, 2020-06-19:
> @tymcauley Hi, may I know how do you access Firesim? Have you purchased the AWS in this case? Thank you very much.

Yes, that's right. I followed the directions [here](https://docs.fires.im/en/latest/) and paid for the necessary AWS instances.

closed 2020-06-19 by tymcauley