firesim/firesim #537: graphml missing at 1.9.0-183-g3c18b391

timsnyder-siv, 2020-04-24:
Hey guys,  I'm digging into the dev branch now that #532 is merged.  I'm noticing that when I run `make  DESIGN=FireSimMulticlockPOC TARGET_CONFIG=DDR3FRFCFSLLC4MB_FireSimQuadRocketMulticlockConfig PLATFORM_CONFIG=F90MHz_BaseF1Config replace-rtl` on commit 3c18b391  (the current tip of dev)

I don't see any `.graphml` file in `sim/generated-src/f1/FireSimMulticlockPOC-DDR3FRFCFSLLC4MB_FireSimQuadRocketMulticlockConfig-F90MHz_BaseF1Config/`

Is this expected?  I'm guessing that it isn't and that's why I'm starting with a GH issue.  If you guys aren't already looking into this, I've been trying to figure it out and will keep trying.

davidbiancolin, 2020-04-24:
It's not expected, though i've never had to look at the graphML stuff before. 

It _should_ have been generated along side the other artefacts in the chipyard generator [here](
https://github.com/ucb-bar/chipyard/blob/dev/generators/firechip/src/main/scala/Generator.scala#L71)

The sources for the old generator utils have been moved into firesim [here](https://github.com/firesim/firesim/blob/dev/sim/midas/src/main/scala/rocketchip/GeneratorUtils.scala)

timsnyder-siv, 2020-04-24:
Thanks @davidbiancolin .  I don't really **have** to look at the graphml.  It's just helping me wrap my head around what the DUT looks like in different elaborated designs.

I'll see if I can figure out how to fix it and post a PR.  Please don't worry about trying to fix this for me.

Hope the writing is going well!

davidbiancolin, 2020-04-24:
If i had to guess, something broke when RC moved to the stage-based generator implementation. The proper fix likely involves porting the generator in chipyard to use stage -- i didn't do that only because there would be a bunch of throwaway work once RC bumps to firrtl 1.3.

Does the graphML show up in other chipyard designs? (If you run `make` in chipyard/sims/verilator)

timsnyder-siv, 2020-04-25:
Yes.  When I run `make -C target-design/chipyart/sims/verilator` in my Firesim clone, I get `target-design/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.graphml`

Note, I'm still on firesim commit 3c18b391

davidbiancolin, 2020-05-04:
This should be resolved by #542, which bumped chipyard for the fix.

closed 2020-05-04 by davidbiancolin