firesim/firesim #555: key not found: plusarg_reader

austinharris, 2020-05-15:
I'm running into the error below when I try to build the firesim simulator with the RV32RocketConfig from chipyard. I saw that there is a PlusArgReaderPass to remove RocketChip's plusarg_reader, but I get a similar error except "plusasrg_reader_0" if I remove that pass. Does anyone have any pointers for tracking this down?

I am on the dev branch of chipyard 7c7b336c3fef8aaf16a7266d95b3fbd9e7d9418b.

`[error] Caused by: java.util.NoSuchElementException: key not found: plusarg_reader
[error]         at scala.collection.MapLike.default(MapLike.scala:232)
[error]         at scala.collection.MapLike.default$(MapLike.scala:231)
[error]         at scala.collection.AbstractMap.default(Map.scala:59)
[error]         at scala.collection.MapLike.apply(MapLike.scala:141)
[error]         at scala.collection.MapLike.apply$(MapLike.scala:140)
[error]         at scala.collection.AbstractMap.apply(Map.scala:59)
[error]         at firrtl.passes.InferTypes$.infer_types_s$1(InferTypes.scala:53)
[error]         at firrtl.passes.InferTypes$.$anonfun$run$14(InferTypes.scala:73)
[error]         at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:234)
[error]         at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:59)
[error]         at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:52)
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at scala.collection.TraversableLike.map(TraversableLike.scala:234)
[error]         at scala.collection.TraversableLike.map$(TraversableLike.scala:227)`

davidbiancolin, 2020-05-31:
Hey Austin. I suspect this has been fixed by #577. If you provide more stack trace i should be able to tell you definitively.

davidbiancolin, 2020-11-09:
Should be resolved in #577.

closed 2020-11-09 by davidbiancolin