firesim/firesim #564: "Setting up your Manager Instance" Docs Correction

hngenc, 2020-05-20:
https://docs.fires.im/en/latest/Initial-Setup/Setting-up-your-Manager-Instance.html

In 2.3.1, we're told to copy-paste a script that gets run when we log into our AWS server. The script contains this line:
`sudo yum -y install https://centos7.iuscommunity.org/ius-release.rpm`

In fact, the link given in this command has apparently been deprecated, and it didn't work for me. The correct link appears to be:
`sudo yum -y install https://repo.ius.io/ius-release-el7.rpm`

abejgonzalez, 2020-05-20:
This is fixed in #563 and will be updated in the next FireSim release.

davidbiancolin, 2020-05-31:
We're leaving this open so that it'll be discoverable for people setting up fresh managers with older FireSim versions.