firesim/firesim #580: Experimental support for ssh into simulated nodes does not work

linkages, 2020-06-05:
I am trying to follow the directions here:

https://docs.fires.im/en/latest/Advanced-Usage/Miscellaneous-Tips.html#experimental-support-for-sshing-into-simulated-nodes-and-accessing-the-internet-from-within-simulations

If I remove the NUMDOWNLINKS and NUMUPLINKS from switchconfig.h as per the directions, make fails with the following:

```
In file included from switch.cc:77:0:
flit.h: In function ‘uint16_t get_port_from_flit(uint64_t, int)’:
flit.h:77:21: error: ‘NUMDOWNLINKS’ was not declared in this scope
     if (sendport == NUMDOWNLINKS) {
                     ^
flit.h:79:32: error: ‘NUMUPLINKS’ was not declared in this scope
         int randval = rand() % NUMUPLINKS;
                                ^
In file included from switch.cc:79:0:
shmemport.h: In constructor ‘ShmemPort::ShmemPort(int, char*, bool)’:
shmemport.h:39:23: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
         recvdirection = "stn";
                       ^
shmemport.h:40:23: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
         senddirection = "nts";
                       ^
shmemport.h:43:23: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
         recvdirection = "nts";
                       ^
shmemport.h:44:23: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
         senddirection = "stn";
                       ^
switch.cc: In function ‘void do_fast_switching()’:
switch.cc:177:29: error: ‘NUMDOWNLINKS’ was not declared in this scope
         for (int i = 0; i < NUMDOWNLINKS + ADDUPLINK; i++) {
                             ^
switch.cc:173:20: error: ‘NUMUPLINKS’ was not declared in this scope
 #define ADDUPLINK (NUMUPLINKS > 0 ? 1 : 0)
                    ^
switch.cc:177:44: note: in expansion of macro ‘ADDUPLINK’
         for (int i = 0; i < NUMDOWNLINKS + ADDUPLINK; i++) {
                                            ^
In file included from switch.cc:247:0:
switchconfig.h: In function ‘int main(int, char**)’:
switchconfig.h:9:142: warning: deprecated conversion from string constant to ‘char*’ [-Wwrite-strings]
     ports[0] = new ShmemPort(0, "0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000", false);
                                                                                                                                              ^
make: *** [switch] Error 1
```
I believe that is because the NUMDOWNLINKS and NUMUPLINKS is switchconfig.h are needed in switch.cc and flit.h.

I went ahead and put them back in switchconfig.h and ran make. This completed succesfully. I then followed the rest of the steps until getting to Step 9. At that point the simulation starts but the screen terminal gets stuck at "Starting network: OK" for a long time.

If I run the same simulation without the changes to the switch, the simulation starts and I can get a terminal on the simulated nodes.

linkages, 2020-06-05:
Here is the contents of config_runtime.ini:

```
# RUNTIME configuration for the FireSim Simulation Manager
# See docs/Advanced-Usage/Manager/Manager-Configuration-Files.rst for
# documentation of all of these params.

[runfarm]
runfarmtag=mainrunfarm

f1_16xlarges=0
m4_16xlarges=0
f1_4xlarges=0
f1_2xlarges=1

runinstancemarket=ondemand
spotinterruptionbehavior=terminate
spotmaxprice=ondemand

[targetconfig]
#Set topology=no_net_config to run without a network simulation
#topology=no_net_config
topology=example_1config
no_net_num_nodes=1
linklatency=6405
switchinglatency=10
netbandwidth=200
profileinterval=-1

# This references a section from config_hwconfigs.ini
# In homogeneous configurations, use this to set the hardware config deployed
# for all simulators
defaulthwconfig=firesim-boom-singlecore-nic-l2-llc4mb-ddr3

[tracing]
enable=no

# Trace output formats. Only enabled if "enable" is set to "yes" above
# 0 = human readable; 1 = binary (compressed raw data); 2 = flamegraph (stack
# unwinding -> Flame Graph)
output_format=0

# Trigger selector.
# 0 = no trigger; 1 = cycle count trigger; 2 = program counter trigger; 3 =
# instruction trigger
selector=1
start=0
end=-1

[autocounter]
readrate=0

[workload]
workloadname=linux-uniform.json
terminateoncompletion=no
suffixtag=

[hostdebug]
# When enabled (=yes), Zeros-out FPGA-attached DRAM before simulations
# begin (takes 2-5 minutes).
# In general, this is not required to produce deterministic simulations on
# target machines running linux. Enable if you observe simulation non-determinism.
zerooutdram=no
```

linkages, 2020-06-05:
After many hours of waiting, the simulated node booted to a login prompt. When I try to login, it hangs after I try to login as the root user via the screen session.

I also note that the switch0 simulation process died at some point but I am not sure when.

daviddpruitt, 2020-08-18:
I have the same problem. Following the instructions the simulated system seems to get stuck. I used 1 uplink and 1 downlink port to try and get it to work. I also had a problem with the line

`ports[0] = new ShmemPort(0)`

It wouldn't compile because it requires 3 parameters so I also used the line from the original. My switchconfig.h file looks like this:

`// THIS FILE IS MACHINE GENERATED. SEE deploy/buildtools/switchmodelconfig.py

#ifdef NUMCLIENTSCONFIG
#define NUMPORTS 2
#define NUMUPLINKS 1
#define NUMDOWNLINKS 1
#endif
#ifdef PORTSETUPCONFIG
ports[0] = new ShmemPort(0, "0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000", false);
ports[1] = new SSHPort(1);
#endif

#ifdef MACPORTSCONFIG
uint16_t mac2port[3]  {1, 2, 0};
#endif`

sagark, 2022-07-06:
This was fixed by #1119, should be visible in main branch's docs now.

closed 2022-07-06 by sagark