firesim/firesim #599: AWS+FireSim issues for long simulation runs (the "lonely" platform)

kiniry, 2020-07-10:
We have built more than one Rocket-based platform using FireSim.  Some are RV32s booting FreeRTOS and others are RV64s booting Linux and FreeBSD.  

We are witnessing a problem with "idle" targets eventually failing.  By "idle" I mean, e.g. a FreeRTOS boot that witnesses zero UART, network, or device interaction, and no debugger interaction, sees a full CPU stop and OS crash after a nearly-deterministic period of time.  But, if we exercise the platform, say by pinging the network stack inducing ICMP relies, or interact with it over UDP or TCP, or keep it busy with a debugger, the platform never crashes.  

Do any FireSim developers or very knowledgable FireSim users have any insights on this strange behavior?   Has anyone witness mysterious, and perhaps undocumented, behavior in AWS that might cause such a "lonely" platform to fail?

davidbiancolin, 2020-07-11:
Hey Kiniry,

That's quite strange. This:
> Sees a full CPU stop and OS crash after a nearly-deterministic period of time
is interesting though.

Approximately how long is this period of time? Is it consistent across different types of target machines (i.e., does it occur after a deterministic amount of simulated time)? Do you have any output from the simulated system at the time it undergoes the full-stop or does it just hang?

I have more questions, but more pressingly, what FireSim version are you using?

kiniry, 2020-07-14:
An update on our end after further testing.  Thanks for the quick response. 

In the end, the problem came down to a bug in the ported to RV32 IP device driver wherein we discovered that the outbound TX queue was only being flushed when a packet was being inserted into the RX queue.  Thus, if your "lonely" CPU/OS was very occasionally sending an ARP packet, it would eventually fail due to a full TX hardware queue (at time A*TXsize where A = ARP periodicity and TXsize is the size of the hardware TX queue; in our case this was 2,950 seconds).  

We also received further information from AWS about undocumented and subtle matters wrt F1 infrastructure that might impact FireSim and/or our applications.

Given the above, I'm closing this issue.  Thanks again @davidbiancolin and the FireSim team.

closed 2020-07-14 by kiniry

davidbiancolin, 2020-07-14:
> We also received further information from AWS about undocumented and subtle matters wrt F1 infrastructure that might impact FireSim and/or our applications. 

Could you share this?

kiniry, 2020-07-14:
Sure, drop me an email and I'll send it to you.  kiniry at galois.com