firesim/firesim #600: Decouple TracerV-enable and Trigger-enable

alonamid, 2020-07-11:
In the current FireSIm version, in order to use any bridge that may depend on a trigger, the TracerV bridge driver needs to be enabled.
I previous versions of FireSim, the trigger could be used in any AGFI that had the TracerV widget synthesized (all of them, by default), without having the trace written out to disk.

As a result, in order to use Autocounter in the current version of FireSim, the user must also output a tracefile (which really slows down simulation, because of writing the file to disk).

The rational behind the change was that TracerV was hurting FMR when it was not being used, since it was still offloading token to the host despite them not being written to disk.
The potential solution is to separate trigger-enable and tracer-enable

davidbiancolin, 2020-11-09:
This is partially resolved in #611.

davidbiancolin, 2020-11-10:
Update docs for 1.11

PR #611 (2020-07-16): Fix TracerV Trigger Generation When Tracing Is Disabled
In a regression introduced by multiclock support, if you weren't collecting an instruction trace with TracerV it's trigger sources would be disabled, which would effectively disable AutoCounter. This PR fixes that by separating trace disable from the trigger specification. 

Resolves #600. Perhaps a better longer term solution might be to pull the trigger source handling out of tracerV entirely.
closed 2022-01-23 by davidbiancolin