firesim/firesim #605: Gracefully Remove Autocounter Annotations When Not Enabled

davidbiancolin, 2020-07-15:
Presently, if AutoCounter is not enabled, AutoCounterAnnotations present in the source will produce renaming errors. It might not be obvious to the user this is being created by an AutoCounterAnnotation if they didn't add it themselves. The compiler should gracefully remove these, perhaps warning the user it is doing so.

davidbiancolin, 2020-11-09:
Alon what you have to do here is:
1) Convert the firrtl autocounter anntotations to use "HasDontTouches" and remove the DontTouchs in the annotator utilities.
See [this trait](https://github.com/freechipsproject/firrtl/blob/ecb96e83324ea17cf38b7b90753d745d3c7f51bd/src/main/scala/firrtl/transforms/OptimizationAnnotations.scala#L19). There's also `DontTouchAllTargets`which may be acceptable.

2) Delete the Autocounter annotations at the end of the transform.

timsnyder-siv, 2020-12-03:
I think that there may also be a transform ordering issue that has to be dealt with.  The renaming error I ran into happens when DeadCodeElimination is run because it wants to remove the dead, unused AutoCounter signals.  The AutoCounter transform currently happens after DeadCodeElimination (see [MidasTransforms.scala](https://github.com/firesim/firesim/blob/659ddbf418d033f3b2fa3edb5bce46479dd0e3be/sim/midas/src/main/scala/midas/passes/MidasTransforms.scala#L39-L50))

This might be easier to do than fixing chipsalliance/firrtl#1997 as a quick way to avoid [the intersection of dead blackboxes and the ILA flow](https://forums.xilinx.com/t5/Synthesis/synth-design-no-longer-removes-unreachable-blackboxes-Synth-8/td-p/1181298).

@alonamid have you started looking at this?  I might be motivated to give it a try (instead of writing a script that will do a lame text delete on the dead blackboxes I know about in my design).

timsnyder-siv, 2020-12-03:
You can observe the assertion in 'make -C sim replace-rtl' using the local-chisel34 branch after reverting 5ccad8ab9bb6f175b4bbd6e0e46b74c21fee31ae  (that commit adds --no-dce to the GodenGate invocation to avoid this issue)

alonamid, 2020-12-03:
@timsnyder-siv , I was planning to start looking at this late next week, but if you have time to give it a try beforehand then go ahead!

DCE ignoring dont_touch seems like a pretty concerning issue to me with respect to all the debugging features which are based on signal insertion during firrtl transforms, but I guess that's the reason we need to switch over to these specific types of annotations.

timsnyder-siv, 2020-12-03:
Looks like @davidbiancolin beat us both with 4ec71df  in #668 !

timsnyder-siv, 2020-12-04:
I can confirm that after cherry picking 4ec71df I no longer hit the AutoCounterAnnotation update assertion complaining about the rename map showing a deletion.  However, I'm now running into Vivado complaining about:
```
[192.168.0.213] out: ERROR: [DRC INBB-3] Black Box Instances: Cell 'WRAPPER_INST/CL/firesim_top/top/sim/target/FireSim/lazyModule_/system/dtm/dtmInfoChain' of type 'CaptureUpdateChain' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
```

This comes from Rocket's [DebugTransport.scala](https://github.com/chipsalliance/rocket-chip/blob/66a60e55a18fae859730b01c04450b2da3630e19/src/main/scala/devices/debug/DebugTransport.scala) and might be a different DCE failure case to report.  I'm going to see if the default config in local-chisel34 also leaves this dead module in the verilog.  You guys might also run into this issue when you use AutoILA.

davidbiancolin, 2020-12-04:
The same design simulates fine in ML sim though?

timsnyder-siv, 2020-12-04:
> The same design simulates fine in ML sim though?

I would assume so.  I'll go double check. 

Vivado is the only thing I've seen complain about trivial dead modules like this.  I don't really understand why it considers them blackboxes either because they have a module definition supplied in the cl_firesim_generated.sv,  it just happens to only have input ports.   

I think there is an intersection with them being left in the RTL and AutoILA preventing Vivado from removing them at the end of `synth_design` as I explained in [my post on the Xilinx Community Forum](https://forums.xilinx.com/t5/Synthesis/synth-design-no-longer-removes-unreachable-blackboxes-Synth-8/td-p/1181298).  I did notice in my build with AutoILA that the 'Cell' has the IS_DEBUGGABLE read-only property set to true.  I'm not sure if that is preventing Vivado from removing it at the end of `synth_design` and if so, why it is being set.

timsnyder-siv, 2020-12-04:
Well, I retract my previous comment at least partially.

After digging around for a few hours trying to understand what the `dtmInfoChain` is and why I'm ending up with CaptureUpdateChain that looks like:
```
  CaptureUpdateChain dtmInfoChain ( // @[DebugTransport.scala 133:29]
      .clock(dtmInfoChain_clock),
      .reset(dtmInfoChain_reset),
      .io_chainIn_capture(dtmInfoChain_io_chainIn_capture)
    );
```

I went back to look at the build I did before cherry-picking 4ec71df and the `CaptureUpdateChain dtmInfoChain` looks like 
```
 CaptureUpdateChain dtmInfoChain ( // @[DebugTransport.scala 133:29]
    .clock(dtmInfoChain_clock),
    .reset(dtmInfoChain_reset),
    .io_chainIn_shift(dtmInfoChain_io_chainIn_shift),
    .io_chainIn_data(dtmInfoChain_io_chainIn_data),
    .io_chainIn_capture(dtmInfoChain_io_chainIn_capture),
    .io_chainIn_update(dtmInfoChain_io_chainIn_update),
    .io_chainOut_shift(dtmInfoChain_io_chainOut_shift),
    .io_chainOut_data(dtmInfoChain_io_chainOut_data),
    .io_chainOut_capture(dtmInfoChain_io_chainOut_capture),
    .io_chainOut_update(dtmInfoChain_io_chainOut_update),
    .io_capture_bits_reserved1(dtmInfoChain_io_capture_bits_reserved1),
    .io_capture_bits_dmireset(dtmInfoChain_io_capture_bits_dmireset),
    .io_capture_bits_reserved0(dtmInfoChain_io_capture_bits_reserved0),
    .io_capture_bits_dmiIdleCycles(dtmInfoChain_io_capture_bits_dmiIdleCycles),
    .io_capture_bits_dmiStatus(dtmInfoChain_io_capture_bits_dmiStatus),
    .io_capture_bits_debugAddrBits(dtmInfoChain_io_capture_bits_debugAddrBits),
    .io_capture_bits_debugVersion(dtmInfoChain_io_capture_bits_debugVersion),
    .io_capture_capture(dtmInfoChain_io_capture_capture),
    .io_update_valid(dtmInfoChain_io_update_valid),
    .io_update_bits_reserved1(dtmInfoChain_io_update_bits_reserved1),
    .io_update_bits_dmireset(dtmInfoChain_io_update_bits_dmireset),
    .io_update_bits_reserved0(dtmInfoChain_io_update_bits_reserved0),
    .io_update_bits_dmiIdleCycles(dtmInfoChain_io_update_bits_dmiIdleCycles),
    .io_update_bits_dmiStatus(dtmInfoChain_io_update_bits_dmiStatus),
    .io_update_bits_debugAddrBits(dtmInfoChain_io_update_bits_debugAddrBits),
    .io_update_bits_debugVersion(dtmInfoChain_io_update_bits_debugVersion),
    .midasAsserts(dtmInfoChain_midasAsserts)
  );
```

I think something is messed-up because I should still see midasAsserts in the dtmInfoChain, right?  I'm going to try reverting the cherry-pick of 4ec71df to make sure I can get back to having dtmInfoChain actually contain outputs.

timsnyder-siv, 2020-12-04:
> The same design simulates fine in ML sim though?

With the cherry-pick, I can confirm that my design still passes run-asm-tests in ML sim.

timsnyder-siv, 2020-12-04:
And reverting the cherry-pick of 4ec71df generates verilog with the dtmInfoChain restored.  I'm going to run GoldenGate with `-ll trace` and then through @jackkoenig's magical Python script to see which transform is gutting the dtmInfoChain when I have 4ec71df

davidbiancolin, 2020-12-05:
I suspect DCE is doing it. the assertions might be getting pruned out
because a clock might be undriven or tied off or something to that effect.

there must be some annotations labelling those three signals that remain
when DCE is enabled as donttouch. Can you poke through the annotations and
figure out what?

timsnyder-siv, 2020-12-05:
Yes.  I'm in the process of confirming that it is DCE and I'll also look to see if the remaining nets are marked dont_touch before filing an update to chipsalliance/firrtl#1997

timsnyder-siv, 2020-12-08:
I looked back into my pre-Chisel3.4 builds and see that Vivado was removing the `dtmInfoChain` with the `[Synth 8-4649]` message I mentioned in https://forums.xilinx.com/t5/Synthesis/synth-design-no-longer-removes-unreachable-blackboxes-Synth-8/td-p/1181298

I guess all of my comments since @davidbiancolin's re-enable of DCE have been really unrelated to this topic.

timsnyder-siv, 2020-12-08:
@davidbiancolin does #668 resolve this issue?  I think it does and setup the link but we can delete it if I'm missing something.

closed 2020-12-14 by davidbiancolin