firesim/firesim #613: unreliable environmental conditions for automated AWS spin-up for FireSim instances

kiniry, 2020-07-17:
For FETT (https://fett.darpa.mil) we are spinning up FireSim automatically via AWS hooks, then interacting with the spun-up SoC+OS via ssh and UART.  Unfortunately, spinning up FireSim on a fresh AWS instance does not work reliably.  We have to attempt to spin it up more than once before it comes up correctly.

Appended below is a characterization of this problem we are having with FireSim.

Joe
—

Since we only run one FPGA at a time, instead of using FireSim manager, we build the binaries and take them and execute them on an F1 instance directly. We have the following scenarios:
1. Execute the binaries in the background inside a boot-hook user-data script: Firesim boots (i.e. we see UART output) fine.
2. Execute the binaries NOT in the background inside a boot-hook user-data script: Doesn't boot. If we attempt to run the tool manually after logging in to the instance, it doesn't run. If we just run it again, it boots fine (3rd time the charm, the first time being by the user-data script).
3. Execute the binaries in the background inside a shell user-data script (starting with #!/bin/bash instead of #cloud-boothook): Same behavior as in 2.
I am including here the "/etc/cloud/cloud.cfg" config file. It seems to us that these scripts/services if they are executed after our tool, it works fine. But if they get executed before, firesim doesn't boot.
```
users:
 - default

disable_root: 1
ssh_pwauth:   0

mount_default_fields: [~, ~, 'auto', 'defaults,nofail,x-systemd.requires=cloud-init.service', '0', '2']
resize_rootfs_tmp: /dev
ssh_deletekeys:   0
ssh_genkeytypes:  ~
syslog_fix_perms: ~

cloud_init_modules:
 - disk_setup
 - migrator
 - bootcmd
 - write-files
 - growpart
 - resizefs
 - set_hostname
 - update_hostname
 - update_etc_hosts
 - rsyslog
 - users-groups
 - ssh

cloud_config_modules:
 - mounts
 - locale
 - set-passwords
 - rh_subscription
 - yum-add-repo
 - package-update-upgrade-install
 - timezone
 - puppet
 - chef
 - salt-minion
 - mcollective
 - disable-ec2-metadata
 - runcmd

cloud_final_modules:
 - rightscale_userdata
 - scripts-per-once
 - scripts-per-boot
 - scripts-per-instance
 - scripts-user
 - ssh-authkey-fingerprints
 - keys-to-console
 - phone-home
 - final-message
 - power-state-change

system_info:
  default_user:
    name: centos
    lock_passwd: true
    gecos: Cloud User
    groups: [wheel, adm, systemd-journal]
    sudo: ["ALL=(ALL) NOPASSWD:ALL"]
    shell: /bin/bash
  distro: rhel
  paths:
    cloud_dir: /var/lib/cloud
    templates_dir: /etc/cloud/templates
  ssh_svcname: sshd

# vim:syntax=yaml
```

-Ramy

davidbiancolin, 2020-07-17:
Before we dive deeper into this -- the manager [does a bunch of prep work ](https://github.com/firesim/firesim/blob/c2d8e3a46e59222e115a1fdaa7267592e1d3c503/deploy/runtools/run_farm.py#L630)in infrasetup before actually invoking the simulator, including clearing the FPGA and unloading and reloading specific kernel modules -- do you mirror that in your user-script?

As this is more in his wheelhouse, maybe @sagark's has some extra insight.

rtadros-Galois, 2020-07-17:
@davidbiancolin Yes we do.

rtadros-Galois, 2020-07-17:
We ran some more tests. The background versus non-background is irrelevant. It's about `#cloud-boothook`

davidbiancolin, 2020-07-17:
> We ran some more tests. The background versus non-background is irrelevant. It's about `#cloud-boothook`

So 1. also does not work.

rtadros-Galois, 2020-07-17:
> > We ran some more tests. The background versus non-background is irrelevant. It's about `#cloud-boothook`
> 
> So 1. also does not work.

No, both 1 and 2 work.

rtadros-Galois, 2020-07-17:
Update:
We found the problem (this bug was found 3 weeks ago, but we finally decided to solve it properly instead of working around it). To run firesim, we run two binaries: `switch0` and `fireSim-F1`. Apparently, without `#cloud-boothook`, and on a fresh instance, `switch0` takes longer to start. Since we never saw that when we were developing, we were just spawning the two processes, one after another. Simply adding an `except` for the flow control between the processes seem to do the trick (Also, a short (as small as 1 second) delay between the processes can work). 

Thanks for your concern. Hope this helps.

closed 2020-07-27 by kiniry