firesim/firesim #620: Firesim Tutorial fails on `firesim infrasetup` while executing task 'infrasetup_node_wrapper'

bradley-evans, 2020-08-07:
I'm walking through the "getting started" guide, per https://docs.fires.im/en/latest/Running-Simulations-Tutorial/Running-a-Single-Node-Simulation.html.

On 3.1.3.2, after following the steps as given, I encounter the following:

```
2020-08-07 23:37:09,797 [main        ] [INFO ]  FireSim Manager. Docs: http://docs.fires.im
Running: infrasetup

2020-08-07 23:37:09,798 [__init__    ] [DEBUG]  {'hwconf_dict': {'firesim-boom-singlecore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7fb2efcb82d8>,
                 'firesim-boom-singlecore-no-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7fb2efcb8290>,
                 'firesim-rocket-quadcore-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7fb2efcb8368>,
                 'firesim-rocket-quadcore-no-nic-l2-llc4mb-ddr3': <runtools.runtime_config.RuntimeHWConfig instance at 0x7fb2efcb83f8>,
                 'firesim-rocket-quadcore-no-nic-l2-llc4mb-ddr3-half-freq-uncore': <runtools.runtime_config.RuntimeHWConfig instance at 0x7fb2efcb83b0>,
                 'firesim-supernode-rocket-singlecore-nic-l2-lbp': <runtools.runtime_config.RuntimeHWConfig instance at 0x7fb2efcb8320>}}
2020-08-07 23:37:09,806 [aws_resource] [DEBUG]  i-0bfae31f2249fc7f8
2020-08-07 23:37:10,046 [aws_resource] [DEBUG]  {'Name': 'firesim-manager'}
2020-08-07 23:37:10,047 [__init__    ] [DEBUG]  {'autocounter_readrate': 0,
 'defaulthwconfig': 'firesim-rocket-quadcore-no-nic-l2-llc4mb-ddr3',
 'f1_16xlarges_requested': 0,
 'f1_2xlarges_requested': 1,
 'f1_4xlarges_requested': 0,
 'linklatency': 6405,
 'm4_16xlarges_requested': 0,
 'netbandwidth': 200,
 'no_net_num_nodes': 1,
 'print_cycle_prefix': True,
 'print_end': '-1',
 'print_start': '0',
 'profileinterval': -1,
 'run_instance_market': 'ondemand',
 'runfarmtag': 'mainrunfarm',
 'spot_interruption_behavior': 'terminate',
 'spot_max_price': 'ondemand',
 'suffixtag': '',
 'switchinglatency': 10,
 'terminateoncompletion': False,
 'topology': 'no_net_config',
 'trace_enable': False,
 'trace_end': '-1',
 'trace_output_format': '0',
 'trace_select': '1',
 'trace_start': '0',
 'workload_name': 'linux-uniform.json',
 'zerooutdram': False}
2020-08-07 23:37:10,052 [get_deploytr] [DEBUG]  Setting deploytriplet by querying the AGFI's description.
2020-08-07 23:37:10,052 [get_afi_for_] [DEBUG]  agfi-0e6e54cd1923411cf
2020-08-07 23:37:10,052 [get_afi_for_] [DEBUG]  None
2020-08-07 23:37:13,099 [get_afi_for_] [DEBUG]  {u'FpgaImages': [{u'UpdateTime': datetime.datetime(2020, 7, 20, 5, 11, 35, tzinfo=tzlocal()), u'Name': 'firesim-rocket-quadcore-no-nic-l2-llc4mb-ddr3', u'Tags': [], u'PciId': {u'SubsystemVendorId': '0xfedd', u'VendorId': '0x1d0f', u'DeviceId': '0xf000', u'SubsystemId': '0x1d51'}, u'FpgaImageGlobalId': 'agfi-0e6e54cd1923411cf', u'Public': True, u'State': {u'Code': 'available'}, u'ShellVersion': '0x04261818', u'OwnerId': '552479941143', u'FpgaImageId': 'afi-01725c670f5c05406', u'CreateTime': datetime.datetime(2020, 7, 20, 4, 31, 4, tzinfo=tzlocal()), u'Description': 'firesim-buildtriplet:FireSim-DDR3FRFCFSLLC4MB_FireSimQuadRocketConfig-F90MHz_BaseF1Config,firesim-deploytriplet:FireSim-DDR3FRFCFSLLC4MB_FireSimQuadRocketConfig-F90MHz_BaseF1Config,firesim-commit:c65159afb19cb9c51e936df05854e17ac61b5686-dirty'}], 'ResponseMetadata': {'RetryAttempts': 0, 'HTTPStatusCode': 200, 'RequestId': 'd31a6b7e-5267-495c-bcd8-9b9273220f05', 'HTTPHeaders': {'x-amzn-requestid': 'd31a6b7e-5267-495c-bcd8-9b9273220f05', 'transfer-encoding': 'chunked', 'vary': 'accept-encoding', 'server': 'AmazonEC2', 'date': 'Fri, 07 Aug 2020 23:37:12 GMT', 'content-type': 'text/xml;charset=UTF-8'}}}
2020-08-07 23:37:13,739 [bind_real_in] [DEBUG]  Using f1.16xlarge instances with IPs:
[]
2020-08-07 23:37:13,739 [bind_real_in] [DEBUG]  Using f1.4xlarge instances with IPs:
[]
2020-08-07 23:37:13,739 [bind_real_in] [DEBUG]  Using f1.2xlarge instances with IPs:
['192.168.4.82']
2020-08-07 23:37:13,739 [bind_real_in] [DEBUG]  Using m4.16xlarge instances with IPs:
[]
2020-08-07 23:37:13,739 [build_fpga_d] [INFO ]  Building FPGA software driver for FireSim-DDR3FRFCFSLLC4MB_FireSimQuadRocketConfig-F90MHz_BaseF1Config
2020-08-07 23:37:13,739 [flush       ] [DEBUG]  [localhost] local: make DESIGN=FireSim TARGET_CONFIG=DDR3FRFCFSLLC4MB_FireSimQuadRocketConfig PLATFORM_CONFIG=F90MHz_BaseF1Config f1
2020-08-07 23:37:15,505 [build_fpga_d] [DEBUG]  [localhost] success: firesim.pem available in ssh-agent
mkdir -p /home/centos/firesim/sim/output/f1/FireSim-DDR3FRFCFSLLC4MB_FireSimQuadRocketConfig-F90MHz_BaseF1Config
2020-08-07 23:37:15,505 [build_fpga_d] [DEBUG]  [localhost] 
2020-08-07 23:37:15,512 [instance_liv] [INFO ]  [192.168.4.82] Checking if host instance is up...
2020-08-07 23:37:15,513 [flush       ] [DEBUG]  [192.168.4.82] run: uname -a
2020-08-07 23:40:01,938 [flush       ] [DEBUG]  [192.168.4.82] out: Linux ip-192-168-4-82.ec2.internal 3.10.0-957.5.1.el7.x86_64 #1 SMP Fri Feb 1 14:54:57 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
2020-08-07 23:40:01,938 [flush       ] [DEBUG]  [192.168.4.82] out: 
2020-08-07 23:40:01,962 [instance_log] [INFO ]  [192.168.4.82] Copying FPGA simulation infrastructure for slot: 0.
2020-08-07 23:40:01,963 [flush       ] [DEBUG]  [192.168.4.82] run: mkdir -p /home/centos/sim_slot_0/rsyncdir/
2020-08-07 23:40:02,249 [flush       ] [DEBUG]  [192.168.4.82] rsync_project: rsync  -pthrvz -L --rsh='ssh  -p 22 -o StrictHostKeyChecking=no ' workloads/linux-uniform/br-base.img centos@192.168.4.82:/home/centos/sim_slot_0/rsyncdir//linux-uniform0-br-base.img
2020-08-07 23:40:02,249 [flush       ] [DEBUG]  [localhost] local: rsync  -pthrvz -L --rsh='ssh  -p 22 -o StrictHostKeyChecking=no ' workloads/linux-uniform/br-base.img centos@192.168.4.82:/home/centos/sim_slot_0/rsyncdir//linux-uniform0-br-base.img
2020-08-07 23:40:02,542 [flush       ] [DEBUG]  Fatal error: local() encountered an error (return code 23) while executing 'rsync  -pthrvz -L --rsh='ssh  -p 22 -o StrictHostKeyChecking=no ' workloads/linux-uniform/br-base.img centos@192.168.4.82:/home/centos/sim_slot_0/rsyncdir//linux-uniform0-br-base.img'
2020-08-07 23:40:02,543 [flush       ] [DEBUG]  Aborting.
2020-08-07 23:40:02,555 [<module>    ] [ERROR]  Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 334, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 282, in main
    globals()[args.task](simconf)
  File "/home/centos/firesim/deploy/firesim", line 95, in infrasetup
    runtime_conf.infrasetup()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 376, in infrasetup
    self.firesim_topology_with_passes.infrasetup_passes(use_mock_instances_for_testing)
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 397, in infrasetup_passes
    execute(infrasetup_node_wrapper, self.run_farm, hosts=all_runfarm_ips)
  File "/usr/lib64/python2.7/site-packages/fabric/tasks.py", line 420, in execute
    error(err)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 358, in error
    return func(message)
  File "/usr/lib64/python2.7/site-packages/fabric/utils.py", line 62, in abort
    raise e
SystemExit: 1
2020-08-07 23:40:02,556 [<module>    ] [INFO ]  The full log of this run is:
/home/centos/firesim/deploy/logs/2020-08-07--23-37-09-infrasetup-NTBVDBZYVVTUMH72.log
```

Can't seem to get any further than this with the instructions as given.

davidbiancolin, 2020-08-07:
You ran this right: 

```
cd firesim/sw/firesim-software
./marshal -v build br-base.json
```

Could you check that the images are there? You should be able to follow `workloads/linux-uniform/br-base.img` which symlinks out to marshal-compiled image.

bradley-evans, 2020-08-09:
I thought I had. Ran through the procedure again and now it works. Thanks.

closed 2020-08-09 by bradley-evans