firesim/firesim #646: Use assembly FAT JARs in Chisel Build

timsnyder-siv, 2020-10-15:
@davidbiancolin , we've discussed this before as a path toward making #509 easier to implement.  There are others over here that would like to see me take a FAT JAR from our world and give it to FireSim instead of letting FireSim's makefiles call SBT.

I'm to the point where I'd like to implement this and submit a PR if you're still open to the idea.

davidbiancolin, 2020-10-15:
I'm open. How are you proposing to do that? 

Spitballing here, the problem as i see it is that there are three non-NOP recipes in the makefile that do:

1) Target elaboration 
2) GG compilation
3) Preprocessing and moving files into the vivado project directory (replace RTL)

(1) and (2) can be farmed out to a fat jar invocation on the remote. (3) can just be moved into a script. 

You can extend the buildafi function to compile your fat JAR. I think this should probably be a recipe in the makefiles anyways. Then you can dry-run invoke Make to get the arguments for steps (1), (2), (3).

timsnyder-siv, 2020-10-16:
You summarized how I was thinking about it.  I was planning add a target for creating the fat Jar.  Something like `firesim/sim/generated-src/f1/assembly.jar` since it is config independent.

Then I was going to make everything else that uses SBT call the fat JAR instead (and try not to break bloop or the other TARGET_PROJECTs in the process).

I had thought that the stuff you labeled target 3 for the preprocessing and moving stuff into the `aws-fpga/hdk/.../cl_firesim` copy should be broken out into a script.

timsnyder-siv, 2020-10-23:
As I'm starting to look into this in more detail and actually implement it, I'm realizing that Bloop doesn't support creation of a FAT jar.  If we make the fat_jar stuff an option, it is going to add a bunch of complexity to the makefiles.  If it isn't an option, I don't see how to keep the Bloop stuff.

@davidbiancolin do you want to see how ugly it is if I make the FAT_JAR stuff an option or do you think it's okay to remove the bloop support or is there a bloop expert that I should be talking to about this?

davidbiancolin, 2020-10-26:
What if the SBT flow just always built a fat jar and invoked that directly, that should streamline this right? I don't see why these two flows can't coexist. 

You could try upgrading to 1.4.0. The biggest problem is that jvm launch and SBT cross-project key resolution takes 10s of seconds, and specifically that hurts running MIDAS examples, where it can double or triple the runtime of small tests.

Perhaps, a compromise is to push the bloop stuff into the midasexamples makefrag but the invocation of GG remains a problem in that case.

timsnyder-siv, 2020-10-26:
I guess I was just looking at the fact that the dependencies of the targets would change if you were using SBT fat-jar or Bloop flow and that seemed like it would be cumbersome.  I'll take a stab at implementing it with keeping the Bloop stuff in place.

davidbiancolin, 2020-10-26:
When enable bloop is set, it injects additional dependencies into recipes that run a scala application. You could do the same for the fat jar but we probably just need to more cleanly specify  / harmonize how we specify what the scala source dependencies are.

timsnyder-siv, 2020-10-28:
@davidbiancolin , I'm starting to work on the manager portion of using the FAT jar and I'm looking in particular at what you mentioned above:
> (3) [replace-rtl] can just be moved into a script.

Looking at the makefile recipes for replace-rtl, I see that [the Vivado Verilog has some custom sed hacks done to it](https://github.com/firesim/firesim/blob/e1149bd301d495c9835274cfbe114b1d8859a000/sim/target-agnostic.mk#L182-L183) that aren't done to the verilog that is left in `sim/generated-src/f1/<triplet>/FPGATop.v`.  Does it make sense why we only do those sed hacks to the Vivado verilog?

davidbiancolin, 2020-10-29:
Given how the verilog emission presently looks, I think:

1) The `RANDOM replacement should be removed. It seems to me that all of the statements that use `RANDOM should be removed since we should not be setting `RANDOMIZE_REG_INIT and friends. (If they are being set, we should unset them.) If for whatever reason vivado still can't handle this, a cleaner fix is to define `RANDOM in vivado synthesis to be `'0 .

2) The fatal hack is still unavoidable, since i think vivado synthesis croaks on fatals for finish codes =/= 0. (Which $fatal implies). Maybe that can be pushed into the FPGA flow, since I think we should leave it as is for ML simulation.

timsnyder-siv, 2020-10-29:
Sounds good.  I'll remove the random hack and move the fatal hack into the scriptified version of the `replace-rtl` target and immediate prereqs that I'm writing.