firesim/firesim #650: source sourceme-f1-manager.sh error

CristinaZN, 2020-10-20:
after i `source sourceme-f1-manager.sh`, it said
-bash: env.sh: No such file or directory
success: firesim.pem available in ssh-agent
but when i do `./marshal -v build workloads/br-base.json`, it said "Please source firesim/sourceme-manager-f1.sh first"

davidbiancolin, 2020-10-20:
It seems like there might have been a problem during build-setup.sh (during which env.sh is generated). There should be a log of that process in the same directory as the script.

CristinaZN, 2020-10-20:
thanks. 
i retry it and successed

CristinaZN, 2020-10-20:
but when i do ./marshal -v build workloads/br-base.json, it STILL said "Please source firesim/sourceme-manager-f1.sh first"

CristinaZN, 2020-10-20:
Submodule 'env' (git://github.com/ucb-bar/riscv-test-env.git) registered for path 'env'
Cloning into 'env'...
Submodule path 'target-design/firechip/rocket-chip/torture/env': checked out 'ce70afbf50a203be04bc326326cfa75831fe7f5d'
Cloning into 'sifive-blocks'...
Submodule path 'target-design/firechip/sifive-blocks': checked out 'e628b3dea190265ad3c3f28582a1e9ae5667da9e'
Cloning into 'testchipip'...
Submodule path 'target-design/firechip/testchipip': checked out '208daac5bd0ea3f85c3f1a0925dec3fafb129451'
Unable to checkout '380c233b43c2de53b0ee15a39e9364d438066b9f' in submodule path 'sim/firrtl'

is that `Unable to checkout` wrong?

davidbiancolin, 2020-10-20:
Likely another transient failure, especially if you are using a release version of FireSim.

davidbiancolin, 2020-11-09:
Any news on this?

closed 2020-11-10 by davidbiancolin

terryjx, 2021-03-23:
I think I'm having the same issue here, which is that:
after I run build-setup.sh, I could not see env.sh generated, and the following modules are either skipped or unabled to be checked out:
`fatal: reference is not a tree: 380c233b43c2de53b0ee15a39e9364d438066b9f
Skipping submodule 'target-design/firechip/riscv-tools'
Skipping submodule 'target-design/firechip/rocket-chip/riscv-tools'
Unable to checkout '380c233b43c2de53b0ee15a39e9364d438066b9f' in submodule path 'sim/firrtl'`

and before doing that I was able to do finish the single node simulation test following v1.6.0 version of firesim official documentation with the original firesim.

I tried to run build-setup.sh on the manager node that I successfully ran original firesim, and on a brand new manager node with only firesim-nvdla cloned. It's the same.

Any idea what could be cause of such issue?  I'm on US-EAST-1 AWS region which is based off N. Virginia.