firesim/firesim #655: Remove copied VerilogMemDelays from tree

albert-magyar, 2020-11-05:
The fix from freechipsproject/firrtl#1908 was pretty important for FireSim, so it got copied into the tree in #639.

This issue is to remind me to delete it when FIRRTL is bumped far enough to include the upstream fix.

davidbiancolin, 2020-11-10:
I think we're on track to have a 1.4 bump before the release. So i'm putting the 1.11 milestone on this.

timsnyder-siv, 2020-12-09:
Does this need to be done as a part of the Bump to FIRRTL 1.4 (#668) or will FAME-5 still work with the locally copied VerilogMemDelays?

davidbiancolin, 2020-12-09:
It needn't be done as part of the ongoing bump. We're going to try to get that in soon now that CI is passing.

timsnyder-siv, 2020-12-10:
Makes sense.  However, it looks like the addition of VerilogMemDelays is a conflict that has to be resolved for #668 to merge see https://github.com/firesim/firesim/pull/668/conflicts

davidbiancolin, 2020-12-14:
Resolved by #668.

closed 2020-12-14 by davidbiancolin