firesim/firesim #659: Docs Missing init-submodules.sh Step

hngenc, 2020-11-15:
https://docs.fires.im/en/latest/Running-Simulations-Tutorial/Running-a-Single-Node-Simulation.html#building-target-software

These instructions need to be changed to:

```
cd firesim/sw/firesim-software
./init-submodules.sh
./marshal -v build br-base.json
```

Otherwise, we get this error:
```Marshal submodule "/home/centos/firesim/target-design/chipyard/software/firemarshal/wlutil/br/buildroot" not initialized. Please run "./init-submodules.sh."```

davidbiancolin, 2020-11-15:
If you could PR this, that would be lovely.

hngenc, 2020-11-25:
I've created one with a very lucky number: https://github.com/firesim/firesim/pull/666

davidbiancolin, 2020-11-25:
Great, thanks @hngenc.

closed 2020-11-25 by davidbiancolin