firesim/firesim #667: copy_back_job_results_from_run errors on non-existant jobinfo.simoutputs entry

timsnyder-siv, 2020-11-25:
https://github.com/firesim/firesim/blob/579361d67dfaa023a5f51c47e09a3bece7b15ba7/deploy/runtools/firesim_topology_elements.py#L297-L316

@davidbiancolin and @sagark , Is there a reason why the above referenced code uses `warn_only()` when copying back `jobinfo.outputs` but does not when copying back `jobinfo.simoutputs`?  The current behavior where the manager blows up and doesn't continue trying to copy back what was found is problematic when you have outputs that the simulator may or may not create.  For example AUTOCOUNTER* files.  When they exist, I'd like them copied back.  I'd also like to not have to modify my workload.json.  

I suppose, I'd like to propose two changes here:
- add `warn_only()` to the copying back of `jobinfo.simoutputs`
- modify AutoCounter so that it always creates at least an AUTOCOUNTER0 file (even if it just says AUTOCOUNTER was disabled in it) so that we don't get warnings about.  

We may not need both changes but I definitely think we need at least one of them.  Thanks.

davidbiancolin, 2020-11-25:
I'd prefer the prior, there's no reason to blow everything up because the file is missing.

timsnyder-siv, 2020-12-03:
Ok.  I'll submit a PR for this.

PR #681 (2020-12-15): warn_only when getting simoutput files
fixes #667
by making behavior the same when get() fails on simoutput as
what currently happens for output from the rootfs
commit 866c4327 (2020-12-15): warn_only on simoutputs copy back (#681)

fixes #667
by making behavior the same when get() fails on simoutput as
what currently happens for output from the rootfs
closed 2021-01-20 by davidbiancolin