firesim/firesim #669: Interest in tar->AGFI as standalone manager command?

timsnyder-siv, 2020-12-03:
I have some builds I kicked off (with a hacked up manager) that finish Vivado fine and fail to copy back to the manager.

I'm breaking out [the logic that submits the tarball for AGFI creation](https://github.com/firesim/firesim/blob/659ddbf418d033f3b2fa3edb5bce46479dd0e3be/deploy/buildtools/buildafi.py#L141-L236) into a function that can continue to be called from `buildafi.aws_build` and also be called from a new manager 'task' function.  Just to reiterate, I'm not proposing any change in behavior for buildafi, only adding a way to do just the tar->AFI portion if something goes wrong during buildafi and you end up with a useful tarball via more manual methods.

One thing that is required for this to work is being able to pass in a cmdline argument to set launch_time  to an existing prefix of one of the results-build/* directories.

I'm not sure how useful this would be for others.  It could possibly be confusing since it's only useful when things go wrong in buildafi and you have to pick up the pieces.  Might be useful if the build fails and then you can manually fix it up from the checkpoints?

davidbiancolin, 2020-12-03:
i think this is useful. I'm assuming you'd specific a subset of builds using the same field in the build_config.ini?

timsnyder-siv, 2020-12-03:
>  I'm assuming you'd specify a subset of builds using the same field in the build_config.ini?

Yes.  That's how I implemented it.  I was thinking that it would be most useful to iterate over multiple configs.  However, I'm wasn't sure how to get Fabric to run the configs in parallel on localhost.  I'm looking that up and I'll push my implementation to a PR.