firesim/firesim #675: Document and test duplicate sections in manager config files

timsnyder-siv, 2020-12-08:
I was surprised that duplicating a hwconfig in config_hwdb.ini didn't throw ConfigParser.DuplicateSectionError until I looked more closely at the source https://github.com/firesim/firesim/blob/659ddbf418d033f3b2fa3edb5bce46479dd0e3be/deploy/runtools/runtime_config.py#L217-L221

Since you aren't using ConfigParser.read() to read in the .ini file, you have slightly different behavior.  Is it expected that the last section in a file with a given name will win?  If so, document that behavior and add python unit tests to make sure it is and stays the case.

timsnyder-siv, 2020-12-08:
@davidbiancolin , this could be an easy issue to add a first set python of unit tests for to get the ball rolling.  I can do this.  Just let me know if you agree that this needs to be documented.

davidbiancolin, 2020-12-08:
Hmm. I think we should throw an exception here. However, I bet this is Sagar's intended behavior. We can talk about this during the FireSim meeting. 

In this is desired, it definitely needs to be documented.

timsnyder-siv, 2020-12-08:
@davidbiancolin Did you guys have a chance to clarify this during your meeting?

davidbiancolin, 2020-12-09:
I forgot and so followed up in the slack. I think this is intended and so
should be reflected in the docs.

There seems to be some desire to have the manager automatically update the
AGFIs now that it catches most typical build failures better.

- David