firesim/firesim #686: Is it possible  to boot linux on BOOM Mega

franktaTian, 2020-12-13:
I built the Boom mega successfully with aws-fpga.But I cannot  run linux,here is the uartlog :
Script started on Sun 13 Dec 2020 12:01:18 PM UTC
AFI PCI  Vendor ID: 0x1d0f, Device ID 0xf000
Using xdma write queue: /dev/xdma0_h2c_0
Using xdma read queue: /dev/xdma0_c2h_0
UART0 is here (stdin/stdout).
command line for program 0. argc=56:
+permissive +mm_relaxFunctionalModel_0=0 +mm_openPagePolicy_0=1 +mm_backendLatency_0=2 +mm_schedulerWindowSize_0=8 +mm_transactionQueueDepth_0=8 +mm_dramTimings_tAL_0=0 +mm_dramTimings_tCAS_0=14 +mm_dramTimings_tCMD_0=1 +mm_dramTimings_tCWD_0=10 +mm_dramTimings_tCCD_0=4 +mm_dramTimings_tFAW_0=25 +mm_dramTimings_tRAS_0=33 +mm_dramTimings_tREFI_0=7800 +mm_dramTimings_tRC_0=47 +mm_dramTimings_tRCD_0=14 +mm_dramTimings_tRFC_0=160 +mm_dramTimings_tRRD_0=8 +mm_dramTimings_tRP_0=14 +mm_dramTimings_tRTP_0=8 +mm_dramTimings_tRTRS_0=2 +mm_dramTimings_tWR_0=15 +mm_dramTimings_tWTR_0=8 +mm_rowAddr_offset_0=18 +mm_rowAddr_mask_0=65535 +mm_rankAddr_offset_0=16 +mm_rankAddr_mask_0=3 +mm_bankAddr_offset_0=13 +mm_bankAddr_mask_0=7 +mm_llc_wayBits_0=3 +mm_llc_setBits_0=12 +mm_llc_blockBits_0=7 +mm_llc_activeMSHRs_0=8 +slotid=0 +profile-interval=-1 +macaddr0=00:12:6D:00:00:02 +blkdev0=br-base.img +niclog0=niclog0 +blkdev-log0=blkdev-log0 +trace-select=1 +trace-start=0 +trace-end=-1 +trace-output-format=0 +dwarf-file-name=br-base-bin-dwarf +autocounter-readrate=0 +autocounter-filename=AUTOCOUNTERFILE +drj_dtb=br-base-bin.dtb +drj_bin=br-base-bin +drj_rom=br-base-bin.rom +print-start=0 +print-end=-1 +linklatency0=6405 +netbw0=200 +shmemportname0=default +permissive-off br-base-bin 
TraceRV 0: Tracing disabled, since +tracefile was not provided.
random min: 0x0, random max: 0xffffffffffffffff
Commencing simulation.
[    0.000000] OF: fdt: No chosen node found, continuing without
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Forcing kernel command line to: console=hvc0 earlycon=sbi
[    0.000000] Linux version 5.7.0-rc3-58539-g5f5fd87b36e2 (centos@ip-192-168-0-239.cn-north-1.compute.internal) (gcc version 9.2.0 (GCC), GNU ld (GNU Binutils) 2.32) #1 SMP Wed Nov 18 03:39:35 UTC 2020
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] initrd not found or empty - disabling initrd
Is there any help?

Yes,I can run the LageBoomConfig successfullya

davidbiancolin, 2020-12-14:
What branch are you on?

franktaTian, 2020-12-14:
Thanks for reply. 
I clone the chipyard and 
./scripts/firesim-setup.sh --fast --library  to setup firesim as "library"
So ,
the chipyard branch is master
and the firesim branch is : (git branch at ./sims/firesim)
* (HEAD detached at c2d8e3a4)
  master

franktaTian, 2020-12-29:
> What branch are you on?

I still cannot boot linux on Boom MEGA. Any help will be appreciated.
In the paper  "SonicBOOM: The 3rd Generation Berkeley Out-of-Order Machine" , there is a description:"The RISC-V kernel page-table initialization code requires careful
insertion of FENCE instructions to synchronize the TLB as the kernel constructs the page-table entries. As high-performance cores
might speculatively issue a page table walk before a newly constructed page-table-entry has been written to the cache, these FENCE
instructions are necessary for maintaining program correctness.
SonicBOOM’s aggressive speculation found a section of the kernel
initialization code where a missing FENCE caused a stale page-tableentry to enter the TLB, resulting in an unrecoverable kernel page
fault. We are working on upstreaming the fix for this issue into the
Linux kernel."
Does it mean someone has to modify this "section of the kernel" to boot linux. Is there any source code? Or we have to use a "special" branch of OS Kernel?
Is this a bug in BOOM Mega?

Thanks again.

jerryz123, 2020-12-29:
You are likely seeing the kernel bug, that has since been fixed. 
You need to apply this patch: 
https://github.com/torvalds/linux/commit/21190b74bcf3a36ebab9a715088c29f59877e1f3

franktaTian, 2020-12-30:
> You are likely seeing the kernel bug, that has since been fixed.
> You need to apply this patch:
> [torvalds/linux@21190b7](https://github.com/torvalds/linux/commit/21190b74bcf3a36ebab9a715088c29f59877e1f3)

Thanks for your help VERY MUCH!
I "patched" the "old" riscv-linux" under firesim(firemarshal) and The Megaboom can boot linux NOW.

closed 2020-12-30 by franktaTian