firesim/firesim #695: Default FPGADebug Annotations & ILA Platform Config Switch

davidbiancolin, 2021-01-11:
Two things:
- It would be nice to annotate some critical signals as FPGADebug by default (check them in). The or-reduction in the assert bridge being a good example. 
- To enable ^ without wasting resources, it would probably be good to put auto ILA on its own PLATFORM_CONFIG switch, and disabled by default. Of our debugging utilities, AutoILA is the only one that does not do this, so being consistent would be good.

timsnyder-siv, 2021-01-12:
Would it be reasonable to get some of this into 1.11 if we restrict the scope to things that are in firesim and chipyard and I do a first cut PR?  Getting critical signals in rocket and boom (maybe other generators) might take more time.  I was thinking of annotating:
* `assertFire`, `assertId`, `q.io.deq.valid`, and `cycles` from [Assert.scala](https://github.com/firesim/firesim/blob/dev/sim/midas/src/main/scala/midas/widgets/Assert.scala#L25-L53)
* DRAM interface.  Would `EgressReq` and `EgressResp` in [dram/Interfaces.scala](https://github.com/firesim/firesim/blob/dev/sim/midas/src/main/scala/midas/models/dram/Interfaces.scala#L44-L57) be the right place to annotate the DRAM interface so that one could at least see memory transactions coming to/from the DUT?
* Something that fleshes out what @davidbiancolin means by "using an AutoILA" in  [Debugging-Hanging-Simulators.rst](https://github.com/firesim/firesim/blame/f1d8f5fc5ade787ce57c265301d1de00654921c3/docs/Advanced-Usage/Debugging-and-Profiling-on-FPGA/Debugging-Hanging-Simulators.rst#L45)  Depending on how much would need to be annotated, maybe this would be too much to have always going into the ILA.  If so, we could explain more in the documentation about what you should annotate when debugging a hang.
* what else in FireSim would be generally useful for debugging purposes?

davidbiancolin, 2021-01-13:
Is there any need to push for release inclusion? 

> Would it be reasonable to get some of this into 1.11 if we restrict the scope to things that are in firesim and chipyard and I do a first cut PR? Getting critical signals in rocket and boom (maybe [...]
- It might be wise to air on the parsimonious side and just do  assertFire? You should already have the assert ID from a previous run no? 

> * DRAM interface.  Would `EgressReq` and `EgressResp` in [dram/Interfaces.scala](https://github.com/firesim/firesim/blob/dev/sim/midas/src/main/scala/midas/models/dram/Interfaces.scala#L44-L57) be [...]

Mmm this is tricky, but yeah, that's a reasonable spot to snoop on transactions. Though i think annotating host_mem_idle, as well as the stall causes would be my suggestion for things that are always annotated. I wrote a little diplomatic utility that injected FPGADebug annos on an AXI4 interface that we could pull in, but i think you'd want these disabled by default. 

> * Something that fleshes out what @davidbiancolin means by "using an AutoILA" in [...]

I think what would be useful for this would be to add  hardware either to the Sim Wrapper, or the default FAME transform, that could essentially blame a channel (or channels) for simulation stall. We could add an MMIO reg that encoded all of the channels that are currently empty (really, concatenate their valids in an known order), and we could ship the list of channels to the driver so that it could provide some insight into the piece of the simulator that locked up first in the event of deadlock. 

> * what else in FireSim would be generally useful for debugging purposes?
Host assertion synthesis would be chill. i'd need to think about this some more.

davidbiancolin, 2021-01-13:
@alonamid How do you feel about the ILA platform config switch.

timsnyder-siv, 2021-01-14:
> Is there any need to push for release inclusion?

When I started my comment, I was thinking that it would be good to at least announce that we're planning to change the default behavior of AutoILA (i.e. that we're planning to remove `WithILATopWiringTransform` from `BaseF1Config`).  If we were going to have some kind of deprecation plan, maybe the first step should be included in the release.

Now that I look at the code more, it might make sense to rename `WithILATopWiringTransform` to `WithAutoILA`.

> I wrote a little diplomatic utility that injected FPGADebug annos on an AXI4 interface that we could pull in, but i think you'd want these disabled by default.

Yes.  I'd like to have the diplomatic utility for annotating an AXI4 interface and I'm fine with it being disabled by default as long as I know how to enable it.

> Host assertion synthesis would be chill. i'd need to think about this some more.

Yeah, I should have said what other signals should we put FpgaDebug annotations on by default.  Wasn't trying to expand the scope of this issue but I'm curious about where you'd want to put host-side assertions.

timsnyder-siv, 2021-01-29:
Looking at this, it seems like removing `WithILATopWiringTransform` from `BaseF1Config` and possibly renaming it may not be the right thing to do.  When I tried removing it, I run into the Makefiles wanting to

https://github.com/firesim/firesim/blob/69edafe325d619518904eab9f9714e9f3ffd140e/sim/target-agnostic.mk#L196-L199

and the ILA files don't exist.  So, it probably makes sense for there to be a new `WithAutoILA` that sets an 'enable' in the `Config` that the `ILATopWiringTransform` pays attention to.  If it's not enabled, then the ILA files should be written out empty (i.e. skip processing the annotations but still create the files).

@alonamid What do you think about changing the default behavior of AutoILA to better match the other debug features?  I'd really like to be able to have `FpgaDebug` annotations in my chisel but also be able to choose whether or not to hook them all up to the ILA in a given build.  It might also be nice to be able to selectively filter them somehow (similar to the AutoCounter coverpoint filtering) but that could be done in a follow-on issue.

timsnyder-siv, 2021-01-29:
I have a patch that adds a `WithAutoILA` and requires it to be mixed into PLATFORM_CONFIG for the ILA to actually be populated.  I'm trying it out locally and then I'll open a PR.

davidbiancolin, 2021-05-26:
For 1.13.0, lets remove the mixin and annotate some baseline signals.

davidbiancolin, 2022-01-18:
Alternative proposal, put a list of signals in the CY docs.

For 1.13.0, remove the mixin without adding new annotations.

PR #913 (2022-01-25): Remove WithAutoILA from default recipes
Users should explicitly mix this in like other debug features. This makes it easier to check in default FPGADebugAnnotations and makes AutoILA consistent with other debug features.

#### Related PRs / Issues

Resolves #695.

#### UI / API Impact

Users must mix in "WithAutoILA_" to their platform configs.

#### Verilog / AGFI Compatibility

None in practice, since we don't tend to include FPGADebugAnnotations by default.

### Contributor Checklist
- [x] Did you set dev as the base branch?
- [x] Did you add Scaladoc to every public function/method?
- [ ]  [... cut]
closed 2022-02-17 by davidbiancolin