firesim/firesim #703: empty FpgaImages while running launchrunfarm

atishp04, 2021-01-22:
Hi,
I am trying to run a single node simulation. I followed the steps described in https://docs.fires.im/en/latest/Running-Simulations-Tutorial/Running-a-Single-Node-Simulation.html.

After running launchfarm, I got the following error. Any suggestions ?

2021-01-21 23:58:02,592 [get_deploytr] [DEBUG]  Setting deploytriplet by querying the AGFI's description.
2021-01-21 23:58:02,592 [get_afi_for_] [DEBUG]  agfi-01be89add09290c16
2021-01-21 23:58:02,592 [get_afi_for_] [DEBUG]  None
2021-01-21 23:58:04,856 [get_afi_for_] [DEBUG]  {u'FpgaImages': [], 'ResponseMetadata': {'RetryAttempts': 0, 'HTTPStatusCode': 200, 'RequestId': '4077b4f0-5af1-492b-b3e5-968bbe687d43', 'HTTPHeaders': {'x-amzn-requestid': '4077b4f0-5af1-492b-b3e5-968bbe687d43', 'transfer-encoding': 'chunked', 'strict-transport-security': 'max-age=31536000; includeSubDomains', 'vary': 'accept-encoding', 'server': 'AmazonEC2', 'cache-control': 'no-cache, no-store', 'date': 'Thu, 21 Jan 2021 23:58:04 GMT', 'content-type': 'text/xml;charset=UTF-8'}}}
2021-01-21 23:58:04,856 [<module>    ] [ERROR]  Fatal error.
Traceback (most recent call last):
  File "/home/centos/firesim/deploy/firesim", line 334, in <module>
    main(args)
  File "/home/centos/firesim/deploy/firesim", line 280, in main
    simconf = RuntimeConfig(args)
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 363, in __init__
    self.innerconf.print_cycle_prefix)
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 67, in __init__
    self.phase_one_passes()
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 360, in phase_one_passes
    self.pass_apply_default_hwconfig()
  File "/home/centos/firesim/deploy/runtools/firesim_topology_with_passes.py", line 291, in pass_apply_default_hwconfig
    server.get_server_hardware_config().get_deploytriplet_for_config()
  File "/home/centos/firesim/deploy/runtools/runtime_config.py", line 48, in get_deploytriplet_for_config
    'firesim-deploytriplet')
  File "/home/centos/firesim/deploy/awstools/afitools.py", line 125, in get_firesim_tagval_for_agfi
    afi_id = get_afi_for_agfi(agfi_id)
  File "/home/centos/firesim/deploy/awstools/afitools.py", line 38, in get_afi_for_agfi
    return result['FpgaImages'][0]['FpgaImageId']
IndexError: list index out of range
2021-01-21 23:58:04,857 [<module>    ] [INFO ]  The full log of this run is:
/home/centos/firesim/deploy/logs/2021-01-21--23-58-02-launchrunfarm-129DEE5UWN14Z07Y.log

davidbiancolin, 2021-01-24:
Resolved -- AGFIs weren't shared.

closed 2021-01-24 by davidbiancolin

yuzeng2333, 2022-04-02:
Hi, I am seeing the exact same problem: with the fpga-image-global-id in the debug log, no FPGA images are found. Could you please explain how to solve this issue? Thanks!

sagark, 2022-04-05:
Do you mean that you are trying to use the exact AGFI referenced in the original post in this issue? If so, it likely no longer exists (this is from Jan 2021). We only preserve pre-built AGFIs for the latest firesim release, older versions may be erased (see comment: https://github.com/firesim/firesim/blob/main/deploy/sample-backup-configs/sample_config_hwdb.ini#L8). If this is the case, I would recommend bumping to the latest firesim version.

If not, could you tell us which firesim version you are using and the AGFI, if it's a public one?

yuzeng2333, 2022-04-05:
Hi Sagar,

Thank you for your help! I have solved this problem by using another
available AGFI.